olikraus/u8g2 #6: Problems when compiling fonts

basilfx, 2016-05-23:
I noticed two problems when compiling U8g2 (x64, Clang, OS X, `-Wall`):

* Font file lines too large (can be worked around by `-Wno-overlength-strings`). This is just an example (of many):

```bash
u8g2/csrc/u8g2_fonts.c:14647:3: error: string literal of length 12689 exceeds maximum length 4095 that ISO C99 compilers are required to support
      [-Werror,-Woverlength-strings]
  "Z\0\3\2\4\5\4\5\6\20\21\0\375\12\375\13\377\1\234\3I\4\330 \6\0\220\247\0!\10\261"
```

* Binary file size increases because all fonts are compiled. I noticed the same problem with U8glib, which I worked around by pointing to the indiviual fonts. Would it be an option to split fonts into single files (again)?

olikraus, 2016-05-24:
Hmm... i had the impression, that the boundary is at 32K.
See the macro U8G2_USE_LARGE_FONTS
what is the maximum length on your system

basilfx, 2016-05-24:
Regarding second point: I think I may have misinterpreted the output of the font size: I initially compiled for native x64 and my binary increased tremendously compared to U8glib. I remembered I had the same issue when I included all the fonts with U8glib. However, for ARM (Cortex M3), the binary size for old code ported to U8g2 seems to be almost identical:

With `U8G2_USE_LARGE_FONTS` (single font included):

```
   text	   data	    bss	    dec	    hex	filename
  24776	    136	   3208	  28120	   6dd8	pkg_u8g2/bin/stk3600/pkg_u8g2.elf
```

Without `U8G2_USE_LARGE_FONTS` (single font included):

```
   text	   data	    bss	    dec	    hex	filename
  24776	    136	   3208	  28120	   6dd8	pkg_u8g2/bin/stk3600/pkg_u8g2.elf
```

Yes, even with clean-compile the files are identical.

olikraus, 2016-05-24:
I missed the second point in your mail. This is a gcc feature but probably also available with clang: unused data is removed. u8glib and u8g2 depend on this.

olikraus, 2016-05-24:
The 	  U8G2_USE_LARGE_FONTS only affects the compilation of individual fonts. But if you do not use one of these fonts, then you will not notice the difference (there is no difference if the font is not compiled or removed at link time).
However the compiler warning can be avoided (see your first point)

olikraus, 2016-05-29:
issue not clear to me. at least for my version of the gcc everything is working.
can this be closed?

basilfx, 2016-05-30:
I can work around it, but maybe Clang is more pendantic than GCC. Found one more resource [here](https://fedoraproject.org/wiki/Understanding_GCC_Warnings_and_Errors#string_length_.27XXX.27_is_greater_than_the_length_.27509.27_ISO_C90_compilers_are_required_to_support).

You can close it if you want.

closed 2016-05-30 by basilfx

olikraus, 2016-05-30:
Thanks for the link. Nice article. To me it appears like this:
I have two options: Fonts can be stored as arrays or strings. In  U8glib i used arrays, but this leads to very long compilation time (due to this problem: https://gcc.gnu.org/wiki/Speedup_areas#Trees).
In U8g2 i have used strings, which are limited to 4096 chars (but must compiler accept more).

Of course i could revert back to arrays (in fact it is a commandline option in the bdf converter), but compilation time in the Arduino IDE (with gcc in general) will be MUCH longer. I had already a couple of users complaining about the long compilation time in u8glib. I assume the average Arduino user will accept a warning if the turn around time is faster...

basilfx, 2016-05-30:
Indeed, that makes much more sense than longer compile times :-)