olikraus/u8g2 #7: Link between u8x8 and u8g2 unclear

basilfx, 2016-05-23:
This night I tried to understand the relation between u8x8 and u8g2. It's not completely clear how both relate. I see that u8g2 setup code refers to u8x8 code (e.g. in `u8g2_d_setup.c`), so that makes me believe that u8x8 code is responsible for interfacing.

I also understand that u8x8 does not use page buffers and write directly to the display in tiles of 8x8 pixels. How does this work for displays that are line based? I want to add a driver for the Sharp LS013B7DH03, which requires you to write one line at a time (128x1 pixels).

Maybe I'm missing a bit of information, can you help me out?

olikraus, 2016-05-24:
Yes, u8g2 builds on u8x8.
U8x8
- contains all low level functions
- implements a character like access. It allows tiles of size 8x8 pixel to be written directly on the screen by applying a "tile" position. 

As you have noticed, not all displays support this kind of tile access. In such a case also the u8x8 end user API can not be used. However u8x8 still serves as low level api for u8g2.

basilfx, 2016-05-24:
Thank you very much for this explanation!

I now understand how all the messages flow. I have implemented a communication driver for I2C displays and I can now control a SSD1306.

Next step will be to add the new display driver for the LS013B7DH03 (already works with U8glib).

I'll keep this one open until I have finished the new driver.

olikraus, 2016-05-28:
I wil close this. We can track progress of the LS013B7DH03 in #8.

closed 2016-05-28 by olikraus