olikraus/u8g2 #8: Add page buffer offsets

basilfx, 2016-05-23:
For the Sharp LS013B7DH03 128x128 display you write individual lines of 128x1 pixels at minimum. The format is 1 address byte (Y coordinate), 16 bytes of pixel data, and a trailing byte (don't care). It's also possible to send multiple lines at once (e.g. one full page).

It looks something like this:

```
00 | 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 | 17
=========================================================
YY | XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX | TR
..
..
..
YY | XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX | TR
```

I was looking for a way to add these extra two bytes to the page buffer, so I can use SPI DMA to write more data without CPU intervention. This will increase memory consumption, but may lower energy consumption. In U8glib I was thinking of something like this:

* Catch `U8G_DEV_MSG_SET_PIXEL`, increase the X coordinate by 8 (so it moves to next byte)
* In `U8G_DEV_MSG_PAGE_NEXT`, set the 1st byte of each line in the page buffer to the correct address, and the 18th byte to the trailing byte (clearPage clears it all). Then use DMA to transfer full page buffer to the LCD.

Unfortunately, this doesn't work, because increasing the X coordinate will result in problems at the borders (drawPixel checks bounds). So I was looking for a way to do this with U8g2, but as far as I can tell, it's not possible. Therefore, would it be possible to add an option to add offsets when writing to the page buffer?

olikraus, 2016-05-24:
hope i will find some more time to answer this later. You have to catch U8X8_MSG_DISPLAY_DRAW_TILE
You will receive 8x128 pixel at once.

olikraus, 2016-05-24:
I think it will be best to make it work first, then later add DMA access. Moreover DMA should be done in the low level com function only (gpio_and_delay_cb member of the u8x8 struct).

olikraus, 2016-05-24:
I have added support for your display.
List of Arduino constructors:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ls913b7dh03-128x128

of course not tested... please let me know the results. All in all timing of the display is very slow. Values do not fit into u8x8 data structures... so lets hope the best. maybe I need to increase the size of some variables. especially cs setup time is very huge.

basilfx, 2016-05-24:
Thanks! I'll continue with the code and test it.

I don't want to be nitpicky, but you've made a typo in the driver's name :-)

olikraus, 2016-05-24:
ups, thanks for the hint --> typo fixed

olikraus, 2016-05-28:
can this be closed?

basilfx, 2016-06-13:
The feature (buffer offset) is still desired, but you can close it if you are not going to implement this :-)

olikraus, 2016-06-14:
Thanks for the pull request. I have merged your code. So even without the "buffer offset" it will work, right? It is not like that that i do not want to implement this "buffer offset" feature, it is more like this that i am not able to test this and that it will have some impact on the overall architecture. In fact i can not estimate the effort if we implement this correctly (memory calculation is done automatically by an external tool)

basilfx, 2016-06-14:
The code I contributed works without the buffer offsets :-)

olikraus, 2016-06-14:
Great, then, i would just leave it like this.

closed 2016-06-20 by olikraus