olikraus/u8g2 #23: LD7032

olikraus, 2016-06-21:
Add support for LD7032 in SPI and I2C mode

http://www.seeedstudio.com/document/pdf/OCELL_use_guide2.pdf

This was also requested Shilpa (e-mail)

sprabhud, 2016-06-21:
Thanks for helping me out with OLED programming in I2C mode.

olikraus, 2016-06-21:
Please note, that nothing is implemented as of now.

sprabhud, 2016-06-21:
Yes, I understand. Thanks for the heads up!

olikraus, 2016-06-21:
i currently implement the LD7032 for u8g2. Maybe i2c is much simpler than i thought. The command byte is just the command byte of the description and no further prefix byte is required.

sprabhud, 2016-06-22:
We managed to get the OLED screen turn-on finally. However, writing on it
is still a problem. Hopefully it should be done soon too.

Sincerely,
Shilpa Prabhudeva
Chief Prototyping officer
Workbench projects, Halasuru
Bangalore 560008
www.museinc.in

olikraus, 2016-06-22:
> We managed to get the OLED screen turn-on finally.
With u8g2? I just did some coding yesterday but did not do any kind of testing.
Hope i can continue today with testing and bug fixing.

sprabhud, 2016-06-22:
Not with u8g2, I mean the format was the same as you did in u8g2. So it was
with reference with your library, but instead of the using a library, we
wrote only few required instructions on the main code and tried to turn it
on.

Sincerely,
Shilpa Prabhudeva
Chief Prototyping officer
Workbench projects, Halasuru
Bangalore 560008
www.museinc.in

olikraus, 2016-06-22:
SPI version for the LD7032 now is implemented for u8g2

olikraus, 2016-06-23:
I did an initial implementation for I2C and LD7032. However it is not tested.
Download:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Examples should have an constructor for I2C with LD7032.
Note: Only U8g2 API is supported, U8x8 API is NOT supported.

sprabhud, 2016-06-23:
Yes, thanks I am trying to implement it. also, thanks for the update.

olikraus, 2016-06-23:
I tried hard to change my own device to I2C, but i failed. I have the seeedstudio 0.5 OLED with LD7032 controller, but PCB and parts are to small for me to resolder this for I2C. So, at the moment, i will not be able to test the I2C software. I guess you are on your own at the moment.

sprabhud, 2016-06-24:
That's all right. Thanks for your help all along, and quick responses. I am
sure I would not have progressed much on the I2C without your help. I will
let you know if there are any progress on this end.

Thanks again,
Sincerely

olikraus, 2016-06-25:
i close this for now. SPI is supported, but I2C still missing.

closed 2016-06-25 by olikraus