olikraus/u8g2 #25: ST7920 Hardware SPI not working correctly

PascalKoe, 2016-06-25:
Hi,
I've just tested the HW SPI with a ST7920 128X64 LCD.
At the start most of the picture is displayed partly correctly but then it starts the get messed up.
Tested own code and example-code
Examples: http://imgur.com/a/Jvxel
Tested on Arduino Uno
Pins: 
RS -> 10
R/W -> 11
E -> 13
(RST -> 8)

Constructer: U8G2_ST7920_128X64_F_HW_SPI lcd(U8G2_R0, 10, U8X8_PIN_NONE);
and U8G2_ST7920_128X64_F_HW_SPI lcd(U8G2_R0, 10, 8);

olikraus, 2016-06-25:
thanks. i have seen similar effects. I will work on this. Problem seems to be, that the SPI speed is to high for the controller.

olikraus, 2016-06-25:
fixed... (hopefully)

closed 2016-06-25 by olikraus

PascalKoe, 2016-06-25:
Thank you very much! Works flawlessly with a FPS increase of 63% compared to U8Glib :D

commit a581844d (2016-06-25): fixed #25, added some extra delay for the ST7920