olikraus/u8g2 #27: None display when init BMP180

luxin88, 2016-07-02:
````c++
#include <U8g2lib.h>
#include <Wire.h>
#include <Adafruit_BMP085.h>

U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);
Adafruit_BMP085 bmp;

char tmp_buf[] = "123456";
char pre_buf[] = "abcdef";

void initBmp(void) {
  if (!bmp.begin()) {
    Serial.println("Could not find a valid BMP085 sensor, check wiring!");
    while (1) {}
  }
}

void setup(void) {
  Serial.begin(9600);
  initBmp();                //when add this line,  oled has no display, but comment this line oled display normal.
  u8g2.begin();
}

void loop(void) {

  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_profont12_mr);
  u8g2.drawStr(0, 20, tmp_buf);
  Serial.println(tmp_buf);
  u8g2.drawStr(0, 30, pre_buf);
  Serial.println(pre_buf);
  u8g2.sendBuffer();

  delay(1000);
}
````

when I call initBmp in setup function, and oled has no display

olikraus, 2016-07-02:
U8glib uses some custom i2c procedures. Maybe try u8g2 lib from IDE lib
Manager.

Oliver

olikraus, 2016-07-02:
i think i commented the wrong issue.
However, is this solved now?

luxin88, 2016-07-03:
No, not solved, I Still Use u8glib do it

olikraus, 2016-07-03:
The example above is for U8g2, so neither U8g2 nor U8glib work?

Oliver

luxin88, 2016-07-03:
u8glib works, u8g2 works too. BUT when I add `bmp.begin();` u8g2 DO NOT WORK, but u8glib works.

thinks

olikraus, 2016-07-03:
ohhhhhhhh

olikraus, 2016-07-28:
The initial example uses sw i2c with hw i2c. This will probably not work. Question to OP not yet answered whether both devices work with HW I2C.

Testato, 2016-08-06:
normally is it possible use a sw i2c library on the hw i2c pins, because the software implementation is a simple bitbanging method so every pin can use it.
(my SoftwareWire library, in the Arduino Manager, is a working examples)

olikraus, 2016-08-07:
Sure, but i guess you can not use SW I2C and HW I2C at the same time. Well at least i have not tested this. Question is still not answer, whether the OLED will work in HW I2C mode.. well of coure if the OLED is a cheep far east OLED without proper ACK response, then HW I2C will not work. 

Not sure how to proceed here. I assume this is not a u8g2 problem.

olikraus, 2016-09-24:
no more information, so closing this...

closed 2016-09-24 by olikraus

mathieu-mp, 2016-11-19:
I met a similar issue running U8g2lib alongside with Adafruit_MCP9808

I was using U8G2_SH1106_128X64_NONAME_1_SW_I2C constructor.
Switching to U8G2_SH1106_128X64_NONAME_1_HW_I2C constructor solved the issue.

Hardware is banggood OLED SSH1106 128x64.

Thanks Oliver, great library !

olikraus, 2016-11-19:
Thanks for the feedback.

Ariffffff, 2017-04-21:
Hi, I was trying to run PT2258 digital volume ic (I2C) with ssd1306, 
and dont know how library sets address, but your comment solved my issue

> 
Will
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);
work?

>

olikraus, 2017-04-21:
It will work, if the oled supports proper I2C ACK. I2c address can be set via an additional command (see the u8g2 reference manual). By default the lowest valid address for the ssd1306 is used.

Ariffffff, 2017-04-21:
hi, thanks, can you kindly link me to "additional command" for address.

I realized one more thing, with this solution the sram is 76-80% used , while 30% before. I have than tested it with U8g (old library ) and it uses 21% only

olikraus, 2017-04-21:
U8g2 will use the official Arduino Wire lib, which consumes a lot of sram. For u8glib I had used my own code.

Ariffffff, 2017-04-21:
yes, I think you have mentioned that in difference between version 1 and 2 , which gave me idea to try it.
just tested with version 1 and every thing is working fine.
thankyou for time and effort