olikraus/u8g2 #29: I'm not sure whether I found a BUG of u8g library.

QiaoJiannan, 2016-07-08:
When I use u8g lib on 256x128 LCD(driver chip is ST75256), After create the ST65256 special initial code, it works. But it could not work at WIDTH=256, So I check source codes under folder clib. Then I find uncomment "#define U8G_16BIT 1" to use larger piexls. But, it still not work!

So I changed all "uint8_t cnt" to "u8g_uint_t cnt" under clib folder. For example, "u8g_WriteSequence(u8g_t *u8g, u8g_dev_t *dev, uint8_t cnt, uint8_t *seq)" change to 
"u8g_WriteSequence(u8g_t *u8g, u8g_dev_t *dev, u8g_uint_t cnt, uint8_t *seq)"  
And changed all "uint8_t arg_val" to "u8g_uint_t arg_val". There are many file changed.

After change of above, the u8g lib could work at 256x128 LCD.

I'm not sure this is a bug or I'm not use it correctly, email me QiaoJiannan@vip.qq.com if need any detail.

olikraus, 2016-07-08:
nice finding. Usually i would have split the WriteSequence into smaller pieces so that i could live with the 256 byte limit. But i guess your changes are ok.

However, i guess you have used the wrong tracker. this is the issue tracker for u8g2, but you refer  to u8glib, which has a different tracker.

All in all, the problem is solved and i can close this, right?

QiaoJiannan, 2016-07-09:
Yes, close is ok. I'm not understanding u8glib very while, maybe I using it in wrong way.

closed 2016-07-09 by QiaoJiannan

olikraus, 2016-07-10:
I guess there is no wrong way. If your approach works, then it's good.