olikraus/u8g2 #30: Support for SSD1322 NHD under Teensy 3.x

christophepersoz, 2016-07-12:
Hello Olivier,

If you can implement this display SSD1322 from New Heaven Display I can perform tests and try to debug the issues for this display with this new library.

Thanks

olikraus, 2016-07-12:
great, i will start to work on this

olikraus, 2016-07-12:
Christophe, what is the exact NHD Display you own?
Oliver

christophepersoz, 2016-07-12:
Good ! Here the link to the display, the reference is ER-OLEDM032. It's a 256x64 pixel display which support gray scale (4 gray level) and support by a SSD1322.
Do not pay attention to the title, because you can wire the display in HW SPI as I do.
[http://www.buydisplay.com/default/serial-oled-module-price-3-2-inch-display-256x64-screens-white-on-black](url).
It's a blazing fast display on Teensy, almost 90fps in BW, and around 70fps in GR :D

olikraus, 2016-07-13:
Somehow the link does not work for me.

christophepersoz, 2016-07-13:
I corrected the link in github, my mistake.

olikraus, 2016-07-14:
I have finished the ssd1322 port. It is ready for testing.
Arduino lib is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Thanks,
Oliver

christophepersoz, 2016-07-18:
Hi Olivier,

I'm currently stuck on an another problem.
I will test u8g2 for ssd1322 as soon as I can. Thanks !

Christophe

christophepersoz, 2016-07-18:
Hi Olivier,

I just made a quick test with the example FPS.ino, here the issues I'm facing to on build:

`/Applications/Arduino.app/Contents/Java/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/libc.a(lib_a-signalr.o): In function `_kill_r':
signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
/Applications/Arduino.app/Contents/Java/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/libc.a(lib_a-signalr.o): In function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
collect2: error: ld returned 1 exit status`

The way I declare the display :
`#define OLED_CS     10
#define OLED_A0     9
#define OLED_RST    8

U8G2_SSD1322_256X64_F_4W_HW_SPI u8g2(U8G2_R0, OLED_CS, OLED_A0, OLED_RST);
`

As you mentioned, I had enabled the 16bits mode on u8g2.h before the build.
Any idea about this problem ?

Thanks.
Christophe

olikraus, 2016-07-18:
Hi
I think this has nothing todo with U8g2. Calls to kill() and getpid() can not be resolved. Both are system functions, which are call from libc in your case. In general, these function should be there in libc. Maybe you need to apply "-lc" twice. Or you have used the wrong c library (-lc but wrong libc.a is used)

Oliver

christophepersoz, 2016-07-18:
Hi,

I tried this morning to make a build switching the board configuration to an Arduino Mega 2560 and the build ran as it should. But it is impossible to make a build on Teensy (whatever the board used).
I think there is an issue with your lib and the Teensy boards. I'm gonna ask some support on Teensy's forum about that.

I keep you in touch.
Christophe

scottosx, 2016-07-19:
I have the same issue as well with u8g2 and the Teensy. I am using a SSD1306 monochrome display. Note that the u8x8 examples compile and work fine with the Teensy and this display. 

I am also following the discussion on the PRJC forum, OP initiated discussion. https://forum.pjrc.com/threads/35439-Build-issue-on-Teensy-3-2-with-U8G2-library 

Paul has some initial thoughts.

christophepersoz, 2016-07-19:
Yes I started that discussion and waiting for an answer from Paul. AVR -> ARM seems not so direct sometimes !
Like you said it seems that he have some ideas...

olikraus, 2016-07-19:
I have one more idea. u8g2 makes use of assert()
Maybe this casues the issues here. I will disable this in the next release, maybe specify -DNDEBUG as an additional commandline option.

christophepersoz, 2016-08-01:
Thanks Oliver, I will get back to you as soon as aI can try that.

olikraus, 2016-08-09:
moved to next milestone for further testing

christophepersoz, 2016-08-09:
Great! I will have a look on it and make some tests next week.

olikraus, 2016-09-28:
Complete rework on the driver. Now works with the NHD 256x64 OLED.
Status:
u8x8: Done
u8g2: Nothing visible in 8 bit mode, 16 Bit mostly ok, but triangle procedure is wrong (graphicstest.ino)

olikraus, 2016-09-29:
fixed 8 bit mode so that at least something is visible. 16 pixel at the left corner are not cleared and can not be used. In 8 bit mode only 240x64 pixel are used.
Fixed also the triangle problem (wrong entry in display info struct)

closed 2016-09-29 by olikraus