olikraus/u8g2 #31: Rising edge of clock

basilfx, 2016-07-12:
The LS013B7DH03 requires SPI clock mode 'second rising', whereas other displays use 'first falling'. 

I presume the field `sck_takeover_edge` in `u8x8_display_info_t` can be used for this, but is there a standardized mapping of values to clock modes?

I can create a PR to fix this, e.g. taking the modes as defined by https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Mode_numbers?

(Yes, the LS013B7DH03 driver has the wrong clock mode, but I did not notice this until I refactored my platform driver.)

olikraus, 2016-07-12:
sure, pull request is fine.

basilfx, 2016-07-12:
I opened #32 to fix the issue with the LS013B7DH03 driver.

For the SPI modes: well, not all are supported right now, but I don't have any ways of testing all changes. Therefore I think the current fix is sufficient and it should be changed when it is actually needed.

I'll close this one.

basilfx, 2016-07-12:
OK, I am reconsidering this, as I confused some terms:

What I now assume:
`sck_takeover_edge` -> clock sampling phase (CPHA)

What I need (LS013B7DH03 uses inverted clock):
`sck_polarity` -> clock polarity (CPOL)

It would waste memory by adding another variable, so I propose to change it to `sck_mode` and map to the values of https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Mode_numbers

I'll create another PR :-)

PR #37 (2016-07-27): Clock takover edge to clock mode
This PR should fix #31. Can you give it a look, and see if you like it?

I don't have any Arduino boards here, but I think it is still compatible. Any chance you can test this?

For my SSD1306 connected via RIOT-OS, it works. Will test the other display tomorrow.
closed 2016-07-27 by olikraus