olikraus/u8g2 #38: ESP8266 HW i2c

Testato, 2016-08-05:
In your  boardlist https://github.com/olikraus/u8g2/wiki/boardlist
there is support for SW and HW i2c on Esp8266.

But the esp do not have hw i2c. It is writed in the datasheet, and if you check the arduino port for esp there is a bitbanging i2c

olikraus, 2016-08-06:
Thanks for the hunt. I should be more precise here

olikraus, 2016-08-07:
updated board list

closed 2016-08-07 by olikraus

Testato, 2016-08-07:
This boardlist ?
https://github.com/olikraus/u8g2/wiki/boardlist

It seems incorrect again to me

olikraus, 2016-08-07:
HW I2C just means that Wire.h is used.

Testato, 2016-08-08:
But you sre leaved OK in the table to hw and to sw i2c.
A people that look your yable think that esp have an hw i2c, that is not true.

Other problem is that wire.h is used also on sw i2c, like on esp arduino core, because the author modified it.

I think that the simplest way is remove wire.h mention, and simply put an N.A. in the ESP hw i2c

olikraus, 2016-08-08:
ok, done.