olikraus/u8g2 #39: Arabic font

olikraus, 2016-08-05:
Dear Oliver
Thanks for your rapid reply

I think for Persian language I need these Unicode 9.0 Ranges of the Arabic Font :
1) Arabic 0600-06FF
2) Arabic Presentation Forms-A FB50-FBB1
3) Arabic Presentation Forms-B FE70-FEFF

Greetings from Iran
Farzad

farzad321554, 2016-08-05:
You're great Oliver
I'm waiting for Ver2.1
And I promise the first code on my lovely SSD1306 OLED will be:

u8g2.print("اولیور به ایران خوش آمدی");     //Means "Welcome to Iran Oliver"     ;)    ;)

olikraus, 2016-08-07:
I have created several arabic fonts:
https://raw.githubusercontent.com/wiki/olikraus/u8g2/fntpic/u8g2_font_cu12_t_arabic.png
https://raw.githubusercontent.com/wiki/olikraus/u8g2/fntpic/u8g2_font_10x20_t_arabic.png
https://raw.githubusercontent.com/wiki/olikraus/u8g2/fntpic/u8g2_font_unifont_t_arabic.png

olikraus, 2016-08-09:
looks ok, but i am not sure because of the writing direction. 

[image]

closed 2016-08-09 by olikraus

farzad321554, 2016-08-09:
Yes, the direction is right to left.
Is'nt there any hope for kerning?

olikraus, 2016-08-09:
U8g2 fonts are derived from the BDF file format. To my knowledge, this BDF file format does not contain any kerning information. So u8g2 font calculation tool chain can not get and handle this information.

The only option what i could imagine, is that we add the kerning information as an extra table manually. Would this be an option?

olikraus, 2016-08-09:
Currently i am thinking on the introduction of a kerning table, which has to be added manually.
Something like this:
{ 
code1, code2, kerning,
code1, code2, kerning,
code1, code2, kerning,
...
}
Probably you have to experiment a little bit and derive the kerning table manually.

What do you think?

olikraus, 2016-08-09:
afm files contain proper kerning, some fonts are mentioned here.
The kerning table could be derived from the atm files:
http://www.nsl.com/k/pdf/afm/