olikraus/u8g2 #41: U8x8: text scaling

Anton-V-K, 2016-08-07:
I could probably miss something in the documentation, but I don't see a text scaling method for U8x8.
I think this feature should be available for text-only functionality (it is available in Adafruit SSD1306 libary and in SSD1306Ascii).
At least twofold scaling should be supported (the default system font is too small in most cases on 128x64 OLED displays).
And threefold scaling would be a great addition also.

olikraus, 2016-08-07:
I guess this is a misunderstanding.
You can not compare u8x8 with adafruit gfx lib. Instead, compare u8g2 with adafruit gfx lib.
Instead u8x8 was written to have something which can be compared with the "homebrew code" from here: https://www.youtube.com/watch?v=lkWZuAnHa2Y
Of course the video is also not fair, because some homebrew code is compared with much more powerfull libs.

Anyhow, the idea of u8x8 is this: some graphics controller have a build in 8x8 mode, where software can sent a 8x8 pixel block to the display. This hardware feature is used by u8x8. Because there is no scaling on the hardware itself (like allowing 16x16 blocks) it also means, that there is no scaling for u8x8. So, u8x8 just supports what is there on the hardware side. 

Instead you could say, scaling is missing in u8g2. But from my perspective adafruit just introduced scaling, because they did not provide proper fonts.However, these fonts are there with u8g2, so no scaling is needed for u8g2.

Does this help?

Anton-V-K, 2016-08-08:
Thanks for the clarification.
I was going to use `u8x8` for texts displaying, and without scaling this library is not the best option in my case, while `u8g2` seems to be a bit heavy for me.
I don't know how the communication with controllers is implemented, but scaling works fine in `SSD1306Ascii` (which is a library for plain text displaying on one specific controller). Since this library is fast, I wonder whether its approach could be reused in `u8x8`.

olikraus, 2016-08-08:
Ok, i checked this:
https://github.com/greiman/SSD1306Ascii/blob/master/SSD1306Ascii/src/SSD1306Ascii.cpp

Let me see what i can do here...

olikraus, 2016-08-09:
I have implemented this.
[image]

 * [ ] Create picture for reference manual
 * [ ] Update reference manual
 * [ ] Create example

olikraus, 2016-08-17:
void u8x8_Draw2x2Glyph(u8x8_t *u8x8, uint8_t x, uint8_t y, uint8_t encoding);
uint8_t u8x8_Draw2x2String(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s);
uint8_t u8x8_Draw2x2UTF8(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s);

void draw2x2Glyph(uint8_t x, uint8_t y, uint8_t encoding)
void draw2x2String(uint8_t x, uint8_t y, const char *s)
void draw2x2UTF8(uint8_t x, uint8_t y, const char *s)

olikraus, 2016-08-17:
no ref man pic required. test passed on test board

closed 2016-08-17 by olikraus