olikraus/u8g2 #42: kerning table (follow up #39)

olikraus, 2016-08-09:
This is a followup of the discussions happened in #39 

Sofar the idea is to introduce a kerning table with an additional command:
u8g2.setKerningTable

Existing kerning information can be found in .afm files.
AFM file format spec is here: 
https://partners.adobe.com/public/developer/en/font/5004.AFM_Spec.pdf
The kerning table could be derived from the atm files:
http://www.nsl.com/k/pdf/afm/

Another option is to calculate the kerning tables in bdfconv itself under certain conditions:
 * not monospaced, not 8x8 font
 * transparent font

Algorithm:
  Combine each glyph with each other.
  i = 0
  Start Loop
    Start with default distance
    Draw glyphs into the same Bitmap 
    Check for overlapping pixel
  While pixel do not overlap, i++, decrease distance, continue with loop
  if i <= user-def'ed-threshold 
   continue with next glyph
  kerning delta = i - user-def'ed-threshold 
  store pair in kerning table

farzad321554, 2016-08-12:
Hi Oliver
Thanks a lot for your attention
Unfortunately I'm not a professional programmer & all these days I was studying to understand the concept of kerning & kerning tables for Persian language (with no success :D   ) 
I found some libraries for GCC or CodeVision but I don't know how to use them in arduino. One of them is "Osama's Lab ks0108" :
https://sites.google.com/site/osamaslab/Home/projects-list/glcd-library/OSL_KS0108.pdf?attredirects=0

This library has arabic font & the map table (kerning table) inside it and also the algorithm to use different glyphs 

Can this library give you the idea of the algorithm?

olikraus, 2016-08-12:
Thanks for the link.
The problem is, that any kerning table is specific to a font. I have looked at the lib you mentioned above, but the table obviously is specific to fonts of that library.
Currently i am thinking of a more general solution. I mean it would be nice to have kerning also to western languages.

farzad321554, 2016-08-12:
Got it dear friend ;)

Is the terrific general version coming soon? If no, Can we have a pre-released version including the specific solutions mentioned above?

olikraus, 2016-08-12:
The u8g2 part would be easy. This means the feature itself is easy to implement.
The problem will be, how to derive the kerning tables for all these fonts.

There will be something soon...

olikraus, 2016-08-15:
some progress here. Current problems include: Kerning table is still too huge (although already compressed). Additionally: Writing direction is wrong.

 * [ ] Introduce special draw string function with writing direction: drawUTF8WithKerning(x,y,writing_dir, kerningtable, string)
 * [ ] allow a minimum kerning value to remove very low kerning tables from the tables

olikraus, 2016-08-15:
[image]

These are the current results.
Kerning is calculated automatically but improvments are only small. Upper "Hello World" is without kerning. Line below is with kerning.
More worse: The kerning tables require 107 KB. This is probably too much for an embedded controller.

olikraus, 2016-08-15:
Currently the function is called:
u8g2_uint_t u8g2_DrawExtendedUTF8(u8g2_t *u8g2, u8g2_uint_t x, u8g2_uint_t y, uint8_t to_left, u8g2_kerning_t *kerning, const char *str);

all in all, i do not think this approach makes sense.

next step is to introduce a manual table...

farzad321554, 2016-08-17:
You are perfect man
I'm waiting hopefully for future

olikraus, 2016-08-20:
this is the current status:
[image]
upper part has no kerning
lower part has manual kerning
idea is now, that the kerning table is created manually

olikraus, 2016-08-20:
there is now a new function:
u8g2_uint_t drawExtUTF8(u8g2_uint_t x, u8g2_uint_t y, uint8_t to_left, const uint16_t *kerning_table, const char *s)

the for the above example this is:
u8g2.drawExtUTF8(&u8g2, x, y+20, 1, my_kerning_table, "سلام جهان");

with
const uint16_t my_kerning_table[] = 
{
	/* first char, second char, gap reduction value */
	0x646, 0x627, 4, /* ن and ا */
	0x627, 0x647, 5, /* ا and ه */
	0x647, 0x62C, 6, /* ه and ج */

	0x645, 0x627, 3, /* م and ا */
	0x627, 0x644, 4, /* ا and ل */
	0x644, 0x633, 2, /* ل and س */
	
	/* this line terminates the table */
	0xffff, 0xffff, 0xffff
};

closed 2016-08-21 by olikraus