olikraus/u8g2 #43: drawBitmap not work (properly)?

C44Supra, 2016-08-09:
I'm using the rook_bitmap example from U8G in combination with drawBitmap but aren't getting the expected output to my SSD1306 OLED screen (full page mode).

This is the specific command I'm using:
u8g2.drawBitmap( 0, 0, 1, 8, rook_bitmap);

I'm getting a single white pixel at the x7 y6 position and every couple of seconds a few other pixels will flicker. I have tried upping the cnt as well as the height vars to a few different variables but so far nothing has worked. I've tried running page buffer mode as well, but got the same results.

olikraus, 2016-08-09:
Can you provide some code?

C44Supra, 2016-08-10:
`#include <Arduino.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, 10, 9, 12, 11, 13);

const uint8_t rook_bitmap[] PROGMEM = {
  0x00,         // 00000000 
  0x55,         // 01010101
  0x7f,          // 01111111
  0x3e,         // 00111110
  0x3e,         // 00111110 
  0x3e,         // 00111110
  0x3e,         // 00111110 
  0x7f           // 01111111
};

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal menory
  u8g2.drawBitmap( 0, 0, 1, 8, rook_bitmap);
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);
}`

That's the entire thing. rook_bitmap is a direct copy from the u8glib reference manual, with the only alteration being the u8g_progmem being changed into a regular progmem (simply because u8g2_progmem doesn't appear to exist).

The rest is an edited example file from the U8G2 Full Page Hello World example.

olikraus, 2016-08-10:
hmm... ok, i have not tested this, but if i remember correctly, drawBitmap does not (yet) support PROGMEM.
So you just have to remove the PROGMEM keyword from the bitmap definition.

C44Supra, 2016-08-11:
Ah, I see, it does work without putting that variable into PROGMEM. For this particular sketch not using progmem does mean an additional 8 bytes of dynamic memory is being used.

olikraus, 2016-08-11:
Yes, i twill eat up 8 more bytes in RAM area. 
Lets treat this issue as a feature request to implement drawBitmapP()

ToDo:
 * [ ] Implement drawBitmapP
 * [ ] Test
 * [ ] Update Wiki

closed 2016-10-01 by olikraus

sandhan26, 2017-04-14:
I am using U8glib2, and i am still not able to get past the rook_bitmap example. I get a single blue pixel, with or without PROGMEM. Secondly drawBitmapP() does not work with u8g2.