olikraus/u8g2 #44: Incorrect pixel placement - esp8266

nottledim, 2016-08-11:
I'm using an esp8266 (esp-01) with a 128x64 oled display SSD1306 with i2c interface.

I'm using u8g2 version 2.1.5 (downloaded yesterday) with Arduino 1.6.6 and the Generic ESP8266 v2 board library.

Running your example page_buffer/HelloWorld for example shows the text "HelloWorld!" followed by half of the letter "H" from the beginning.  What I've noticed is that the first few pixel columns, 8 I think,  of text placed at the start of a line is duplicated at the end of the line. This happens irrespective of the length of the text or size of the font.

[image]

I've tried various variations including frame buffer, which makes no difference, and different drivers which changes the symptoms but is still wrong.

I use this constructor in your example program:
#define gpio_SDA 0
#define gpio_SCL 2
U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ gpio_SCL, /* data=*/ gpio_SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display

Is there anything I can do to cure this problem?

Thanks

olikraus, 2016-08-11:
What is the display resolution? Is it 128x64 or is it 132x64? Are you ablet to count the horizontal resolution?
You could also try a different chipset. You display might contain the SH1106 controller instead of the SSD1306.

Note: Information from the seller is sometimes not correct...

nottledim, 2016-08-11:
it's 128x64.  I've tried SH1106 controller - with HelloWorld it looks the same i.e. same error.  

I can place text exactly where I want it except that the last few pixels of the line are corrupted.  So I don't think it's a dimension error.  Anyway it all works fine with u8glib with arduino processor.

I've got 2 similar displays and they both show same error.

It's weird isn't it :-)

olikraus, 2016-08-12:
strange...

Mark-Wills, 2016-08-12:
Just to let you know, I'm having the exact same problem with the exact same display board. U8glib and u8x8 in g2lib don't seem to be a problem. It's just a problem with U8G2.

```c++
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, 19,18);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    //u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.setFont(u8g2_font_7x14B_tf);
    u8g2.drawStr(0,10,"Hello!");
  } while ( u8g2.nextPage() );
  delay(1000);
}
```
The above will re-produce the problem. I'm using an Arduino UNO. I tried both hardware and software I2C constructors; same result.

[image]

olikraus, 2016-08-12:
Can you draw some lines like this:

```
void loop(void) {
  u8g2.firstPage();
  do {
    //u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.setFont(u8g2_font_7x14B_tf);
    u8g2.drawStr(0,10,"Hello!");
    u8g2.drawLine(0,0,63,63);
    u8g2.drawLine(63,63,63+63,63-63);
    u8g2.drawPixel(126,20);
    u8g2.drawPixel(127,21);
  } while ( u8g2.nextPage() );
  delay(1000);
}
```

Can you send me the result?

Mark-Wills, 2016-08-12:
Sure. Give me a minute... on it now...

olikraus, 2016-08-12:
wow, thats quick :-)

olikraus, 2016-08-12:
I personally think that this is not a SSD1306 controller... but let us see...

Mark-Wills, 2016-08-12:
Here you go:

[image]

olikraus, 2016-08-12:
hmmm, this is strange...
did you include the drawPixel() commands?
they are not visible, right?

Mark-Wills, 2016-08-12:
Hang on... I'm posting another pic (takes a while to bluetooth it from my phone and re-size it ;-) )

olikraus, 2016-08-12:
yes, a better close up of the upper right part would be interesting

Mark-Wills, 2016-08-12:
Here you go... 
[image]

Mark-Wills, 2016-08-12:
So, the above pic is with the `u8g2.drawStr(0,10,"Hello!");` commented out.

olikraus, 2016-08-12:
yes, ... so... no pixel... right?

olikraus, 2016-08-12:
ok, this "is" strange
let me see whether I can verify this

Mark-Wills, 2016-08-12:
I'm just trying another test....

Mark-Wills, 2016-08-12:
Okay, this shows the problem beautifully...
```c++
int x;

void loop(void) {
  u8g2.firstPage();
  do {
    //u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.setFont(u8g2_font_7x14B_tf);
    u8g2.drawPixel(x,20);
    u8g2.setCursor(10,10);
    u8g2.print(x);
  } while ( u8g2.nextPage() );
  delay(100);
  if(x++>128) x=0;
}
```
I'll see if I can upload a video... Might take a while...

olikraus, 2016-08-12:
ok, i can reproduce the problem
[image]

Mark-Wills, 2016-08-12:
Yeah, you got it. Here's a video:
[https://youtu.be/5wb4CEAcEBo](https://youtu.be/5wb4CEAcEBo)
It's unlisted.

Mark-Wills, 2016-08-12:
So columns 0 to 7 are being copied into columns 120 to 127.

olikraus, 2016-08-12:
yes exactly. Nice video

nottledim, 2016-08-12:
On 08/12/16 20:00, olikraus wrote:
> yes exactly. Nice video
>
>
With that video do you see a rogue pixel at start 5 and 6 I think, placed incorrectly below the main line.

Mark-Wills, 2016-08-12:
No, I'm not seeing that. Might just be a compression artefact on the video. I'm just seeing the pixel from 0 to 7 being copied to 120 to 127. I guess a modulo error ;-)

olikraus, 2016-08-12:
... a programming error.... introduce with the last release...
It is now fixed:
https://github.com/olikraus/u8g2/commit/bb19b64e2d87a217e1e2aa41d38b746825e13cff#diff-e486df1652b0f6e504a83b07d15e02bc

olikraus, 2016-08-12:
The fixed version is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Download this zip and manually install this lib via IDE library manager (i think it is calld "add library as zip" or something like this)

Thank you so much for pointing out the bug. 
@nottledim: please excause the delay on this.

Mark-Wills, 2016-08-12:
Wow! How's that for service ;-)

How do I get the latest files? I'm new to this Github thing...

olikraus, 2016-08-12:
Just click on 
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Mark-Wills, 2016-08-12:
Ah, I see you just posted. Thanks **SO** much for your work on this!

olikraus, 2016-08-12:
Thank you for pointing out this bug!

olikraus, 2016-08-12:
Maybe you can just confirm that this fix works for you.

Mark-Wills, 2016-08-12:
Okay, confirmed working using the little "walking pixel" code above. Thanks again.

olikraus, 2016-08-12:
Great. Then i will close this issue.

closed 2016-08-12 by olikraus

nottledim, 2016-08-12:
On 08/12/16 20:16, olikraus wrote:
> The fixed version is here:
> https://github.com/olikraus/U8g2_Arduino/archive/master.zip Download this
> zip and manually install this lib via IDE library manager (i think it is
> calld "add library as [...]

I did the easy bit; thank you for fixing it (not to mention writing the
library in the first place).

nottledim, 2016-08-12:
On 08/12/16 20:19, olikraus wrote:
> Maybe you can just confirm that this fix works for you.
> 

Yes, that works.  I just tried my app again and it works as expected.

Many thanks

olikraus, 2016-08-12:
Excellent. Thanks for the feedback.