olikraus/u8g2 #46: getStrWidth sometimes wrong

nottledim, 2016-08-13:
I'm trying right justify some text by subtracting the string width from right-hand column location.  There's a few odd things going on.

Using u8g2_font_6x10_tr the wrong length is returned for strings containing "|" or ">".  The function returns length = 1 and length = 4 respectively whereas it should perhaps be 2 for "|" and definitely 5 for ">".

Using u8g2_font_helvB08_tf there is no problem with ">" - it returns length = 4 which in this case is correct but it gets "|" wrong.  So strings "-|" and "lazydog|" for example are given the wrong length.

There's endless possibilities to test and I've only tried one or two.  Here's snippet that can be used to test.

  char *list[] = {"|", ">", "-|", "lazydog|", "lazydog!", "lazy|dog"};
  u8g2_uint_t fh, w;
  u8g2.drawLine(120, 0, 120, 63);

  u8g2.setFont(u8g2_font_6x10_tr);
  //u8g2.setFont(u8g2_font_helvB08_tf);
  u8g2.setFontPosTop();
  fh = u8g2.getFontAscent() - u8g2.getFontDescent();
    
  int i;
  for (i = 0; i < sizeof(list)/sizeof(char*); i++) {
    w = u8g2.getStrWidth(list[i]);
    u8g2.drawStr(120 - w, fh * i, list[i]);
    u8g2.setCursor(0,fh * i);
    u8g2.print(w);
  }

N.B this is not a big deal for me - it's just something I noticed in passing.

olikraus, 2016-08-15:
Thanks.
I will have a look at these issues.

olikraus, 2016-08-15:
this is fixed, however note, that the current string width calculation is still not 100% perfect as the initial offset is part of the returned width.
But it will work for the above example.

closed 2016-08-15 by olikraus