olikraus/u8g2 #53: ESP8266 and HW SPI

olikraus, 2016-08-21:
Currently there are two issues
- It is not working at all (no signal output in the Feather Wing)
- SPI Modes 2 and 3 are swapped in the board package: esp8266/Arduino#2416

olikraus, 2016-08-21:
1. The first problem is solved. Problem was the use of wrong I/O lines which might crash the ESP8266
2. SPI Mode problem is addressed

dratini0, 2017-09-26:
For anyone else bumping into this again, at the time of writing, a pull request (esp8266/Arduino#2418) has been submitted to esp8266/Arduino, but has not yet been merged. Therefore, point 2 still stands.

Misiu, 2019-05-13:
@dratini0 @olikraus this has been fixed in https://github.com/esp8266/Arduino/pull/5948 and released in [2.5.1](https://github.com/esp8266/Arduino/releases/tag/2.5.1)

Misiu, 2019-05-13:
@olikraus could You test this? I've ordered the display (ST7920 based 128*64) but didn't get it yet so can't help.

olikraus, 2019-05-13:
I can try to find a matching hardware. (ToDo: Test)

olikraus, 2019-05-19:
I did some tests and can confirm that the fix works: The ST7920 display works without modification of u8g2.

Hardware:
 - Feather HUZZAH ESP8266
 - 74HC07 as 3.3V to 5V level shifter
 - ST7920 display

IDE: 
  - Arduino 1.8.4
  - esp8266/Arduino  [2.5.1](https://github.com/esp8266/Arduino/releases/tag/2.5.1)

Test A: Software SPI
```
U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 13, /* CS=*/ 15, /* reset=*/ 16);
```
Works as expected

Test B: Hardware SPI (new code)
```
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, /* CS=*/ 15, /* reset=*/ 16);
```
Now also works.

Test C: Hardware SPI (new code removed)
Same example as Test B: As expected: Example does not work any more.
```
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, /* CS=*/ 15, /* reset=*/ 16);
```

Conclusion: All tests are passed.

olikraus, 2019-05-19:
Testenvironment: Adafruit ESP8266 board,74HC07 and ST7920 display

[image]

closed 2019-05-19 by olikraus

Misiu, 2019-05-23:
@olikraus sorry for the noob'ish question, but why do You need 74HC07? Based on what I found I think ST7920 can be directly connected to ESP8266 or Arduino (for example https://www.hackster.io/user0035382/adjusting-clock-on-128x64-monochrome-display-st7920-b9569a).
I'm still waiting for the display I ordered so I'll check that when it arrives, but I don't want to damage it at the start.

If I want to use Hardware SPI (which is the fastest way right?) I must use `U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, /* CS=*/ 15, /* reset=*/ 16);`
but what pins I must connect? I'll be using Wemos D1.

Thanks for the advice 🙂

olikraus, 2019-05-23:
The hackster example makes use of an Arduino Nano, which is a 5V board. The ESP8266 however is a 3.3V board. For a 3.3V board, the maximum output voltage of the GPIO lines are 3.3V (maybe little bit lesser). But a 5V display (the ST7920 operates with 5V logic) expects at least 3.5V to detect a logic high. In other words, whatever the ESP8266 sends, will be read as logic zero. To avoid this we have to do voltage level translation: The maximum output voltage has to be increased. There are many options to do this, but I just looked for an option based on the available parts in my lab. So I took a 74HC07 device which has an open collector output. Together with a 2K resistor for each signal line  it is a good 3.3V to 5V level translator.

Misiu, 2019-05-24:
@olikraus and that's why this was a noob'ish question. I know that ESP8266 has 3.3V logic, but I wasn't aware that ST7920 has 5V logic. I've searched a bit, found a tutorial with Arduino and assumed that it will work the same way with ESP.

Thanks for clarification

olikraus, 2019-05-24:
> will work the same way with ESP.

Actually, it may work (3.3V is not that far away from 3.5V) but it will not be reliable.

Misiu, 2019-05-24:
@olikraus I'll check that when my display arrives :) 
Nevertheless, I'm going to order a couple of level shifters, just in case.

Kroshtandt, 2020-05-09:
@olikraus hello, have the same problem with nodemcu and ST7920 spi connection. Display is blank. Do I need to use level converter for that?

olikraus, 2020-05-10:
Discussion continued at #1186