olikraus/u8g2 #56: Feature request - XOR for fontMode

Mark-Wills, 2016-08-26:
Hi there. My apologies for posting this as an "issue" - to be clear, this is not a bug, it is a feature request. I did look for a section on posting feature requests but didn't see anything, so my apologies if this is posted in the wrong place.

I am wondering how easy/difficult it would be for you to add an XOR draw mode to fontMode. Currently, fontMode can plot 0's or 1's, but it would be GREAT if it could plot in a XOR mode, too. Then we could write text on top of (for example) shapes.

For my application, I have a numeric BPM (beats per minute) indicator. It uses a large font because the unit will be floor based, so a large font is required in order to be able to see it. I have also square, sawtooth, sine, and triangle waveform modes. I would like to draw the appropriate waveform on the bottom half of the screen as a solid (filled to the bottom edge of the screen) waveform. Then, the BPM value will be drawn, partially on top of the waveform shape. So, the top part of the BPM value will be in "empty" screen space, but the bottom half will be on top of the waveform. In order to pull this off, I'll need an XOR draw mode.

Obviously, this will mean that some portions of the BPM numeric value are inverted, but the human brain will be able to read it just fine. I think it will look great!

Since the library uses a RAM based buffer, I presume it's quite simple to do, but I must admit I have not studied the source code yet myself.

What are your thoughts?

Many thanks

Mark

olikraus, 2016-08-29:
Adding feature requests here is fine.
Your request could be implemented in line 410 of u8g2_font.c
Actually fonts are finally drawn by a horizontal line procedure. This procedure has to do the pixel XOR.

I do hesitate here a little bit, because this part is a little bit speed critical. And more over some other topics seem to be more urgent.

septillion-git, 2017-01-07:
I would really like this feature as well :) That's a thing I miss compared to Adafruit SSD1306 library :)

olikraus, 2017-01-07:
new draw color "2" for xor mode

| Font Mode | Draw Color | Effect |
|------|------|------|
| solid | 0 | bg color 1, fg color 0 |
| solid | 1 | bg color 0, fd color 1 |
| solid | 2 | bg color 0, foreground is xor'ed |
| transparent | 0 | fg color 0 |
| transparent | 1 | fg color 1 |
| transparent | 2 | fg is xor'ed |

hope this is ok.

septillion-git, 2017-01-07:
Wow, that's fast! That's brilliant! Thank!!

olikraus, 2017-01-07:
todo

 * [ ] ST7920 test
 * [ ] Documentation
 * [ ] Extend graphics test?

olikraus, 2017-01-08:
done

closed 2017-01-08 by olikraus