olikraus/u8g2 #58: More Fonts..

olikraus, 2016-08-27:
Monospace Terminal:
http://www.donationcoder.com/Software/Jibz/Dina/index.html   4 sizes
http://font.gohu.org/   2 sizes
http://upperbounds.net/ or http://www.proggyfonts.net/  several types and sizes
https://github.com/lucy/tewi-font 1 size with several variants

toyo, 2016-09-04:
I hope to support these fonts to use non chinese characters.
http://openlab.ring.gr.jp/efont/unicode/index.html.en

olikraus, 2016-09-04:
There are severl bdf files. Each of them contain 20000 to 30000 glyphs. This is too much for an embedded controller. To work with a font, the number of glyphs should be lesser than 8000 (or better lesser than 2000).

Can you be more specific:Which unicode glyphs should be included? There could be different list for different languages.
Which fonts should be included?

toyo, 2016-09-04:
In Japanese (because I'm Japanese:), 

JIS X 0201 characters are the most important characters where the environment is limited by 8bit.
ftp://unicode.org/Public/MAPPINGS/OBSOLETE/EASTASIA/JIS/JIS0201.TXT
These were incompatible with ISO8859 (latin-1 etc).
Because of these characters is covers only Japanese phonogram,
there are only use the limited (memory) and fixed character width environment
such as HD44780 compatible controller IC.

JIS X 0208 characters are the most important characters for the information exchange now.
ftp://unicode.org/Public/MAPPINGS/OBSOLETE/EASTASIA/JIS/JIS0208.TXT
JIS Code 0x21xx - 0x4fxx are the JIS first level characters which is very important characters.
JIS Code 0x50xx - 0x74xx are the JIS second level characters which is important characters.
Although these classification was obsolete (by 1990, maybe), you can classify them if you want to limit the characters about 3000 or not, I think.

JIS X 0212 characters are extended characters and overwrapped with 7bit+7bit space.
ftp://unicode.org/Public/MAPPINGS/OBSOLETE/EASTASIA/JIS/JIS0212.TXT
I think these are not so important characters.

Please ask me if you have any questions.

olikraus, 2016-09-09:
ok, lets see... i have reduced this to some 3700 glyphs
[JIS0208_u8g2_reduced.txt](https://github.com/olikraus/u8g2/files/464742/JIS0208_u8g2_reduced.txt)

toyo, 2016-09-10:
Tested, but cannot compile fonts.c for pro micro (atmega32u4).
--
C:\Users\toyo\Documents\Arduino\libraries\U8g2\src\clib\u8g2_fonts.c:66925:15: error: size of array 'u8g2_font_jis0208' is too large
--

Also 8266, linker causes an error.
--
"section `.rodata' will not fit in region `dram0_0_seg'"
--

Need to do more small?

olikraus, 2016-09-10:
well, yes, thats what i try to say. For small uC the number of glyphs is still to high.
I have not even tried to test this.

toyo, 2016-09-16:
Through my survey, I understand that almost ideogram characters which is used by Japanese newspapers is in u8g2/tools/font/bdfmaps/joyo-kanji-japanese.map, but it is still large to use.

# To check joyo-kanji, use http://joyokanji.info/ (only for Japanese..).

Hmm...

olikraus, 2016-09-19:
ok, i found this: http://www.tonypottier.info/Unicode_And_Japanese_Kanji/
But probably more important is appendix 2:
http://www.tonypottier.info/Unicode_And_Japanese_Kanji/Appendix2.html
could this be a suitable list?
I think, this includes Jōyō kanji (1..6)
Along with this list, there should be  U+3040–U+309F (Hiragana) and  U+30A0–U+30FF (Katakana).

olikraus, 2016-09-19:
ok, here are two mapping tables. First of the two is Joyo kanji.

[japanese1.txt](https://github.com/olikraus/u8g2/files/481212/japanese1.txt)
[japanese2.txt](https://github.com/olikraus/u8g2/files/481213/japanese2.txt)

toyo, 2016-09-19:
Thanks.

Joyo Kanji (1..6) is called Kyoiku Kanji (Educational Kanji), which the students of elementary school (6 years) learn. I tried to use Educational Kanji set last week, however it is not enough to show the almost Internet messages I think.

Anyway, I'll try to use your mapping tables later.

olikraus, 2016-10-01:
closing this...

closed 2016-10-01 by olikraus