olikraus/u8g2 #60: Weird behaviour drawXBM

bvlet, 2016-09-16:
Hi,
I've been struggling all day now to get a logo on my SH1106 OLED. The controller is an ESP8266 and everything is working perfectly, fonts, drawing, etc, but the bitmap is not. I've tried several bitmap 2 hex generators and also Gimp, but the image is always messed up. It looks like the bitmap is put on the screen several times at once and sometimes also flipped:

[http://imgur.com/a/B5VwR]

The code I use for the BMP is this and should write 'FESTO", but instead I get this gibberish...:

`const unsigned char FESTO[330]= {
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x7f, 0xff, 0xfc, 0xff, 0xff, 0xfe, 0x7f, 0xff, 0xfc, 0xff, 0xff, 0xfe, 0x3f, 0xff, 0xfc,
  0x7f, 0xff, 0xfc, 0xff, 0xff, 0xfe, 0xff, 0xff, 0xfc, 0xff, 0xff, 0xfe, 0x7f, 0xff, 0xfc,
  0x7f, 0xff, 0xfc, 0xff, 0xff, 0xfe, 0xff, 0xff, 0xfe, 0xff, 0xff, 0xfe, 0x7f, 0xff, 0xfe,
  0x7f, 0xff, 0xfc, 0xff, 0xff, 0xfc, 0xff, 0xff, 0xfe, 0x7f, 0xff, 0xfe, 0x7f, 0xff, 0xfe,
  0x7e, 0x00, 0x00, 0xfc, 0x00, 0x00, 0xfc, 0x00, 0x7e, 0x00, 0x7e, 0x00, 0x7c, 0x00, 0x7e,
  0x7e, 0x00, 0x00, 0xf8, 0x00, 0x00, 0xfc, 0x00, 0x00, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7e, 0x00, 0x00, 0xf8, 0x00, 0x00, 0xfc, 0x00, 0x00, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7e, 0x00, 0x00, 0xfc, 0x00, 0x00, 0xfc, 0x00, 0x00, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7f, 0xff, 0xf8, 0xff, 0xff, 0xf8, 0xff, 0xff, 0xf8, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7f, 0xff, 0xf8, 0xff, 0xff, 0xf8, 0xff, 0xff, 0xfc, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7f, 0xff, 0xf8, 0xff, 0xff, 0xf8, 0x7f, 0xff, 0xfe, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7f, 0xff, 0xf8, 0xff, 0xff, 0xf8, 0x3f, 0xff, 0xfe, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7e, 0x00, 0x00, 0xfc, 0x00, 0x00, 0x00, 0x00, 0xfe, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7e, 0x00, 0x00, 0xf8, 0x00, 0x00, 0x00, 0x00, 0x7e, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7e, 0x00, 0x00, 0xf8, 0x00, 0x00, 0x00, 0x00, 0x7e, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7e, 0x00, 0x00, 0xf8, 0x00, 0x00, 0xfc, 0x00, 0x7e, 0x00, 0x7c, 0x00, 0x7c, 0x00, 0x3e,
  0x7e, 0x00, 0x00, 0xff, 0xff, 0xfc, 0xff, 0xff, 0xfe, 0x00, 0x7c, 0x00, 0x7f, 0xff, 0xfe,
  0x7e, 0x00, 0x00, 0xff, 0xff, 0xfe, 0xff, 0xff, 0xfe, 0x00, 0x7c, 0x00, 0x7f, 0xff, 0xfe,
  0x7e, 0x00, 0x00, 0xff, 0xff, 0xfe, 0xff, 0xff, 0xfc, 0x00, 0x7c, 0x00, 0x7f, 0xff, 0xfc,
  0x7e, 0x00, 0x00, 0xff, 0xff, 0xfe, 0x7f, 0xff, 0xfc, 0x00, 0x7c, 0x00, 0x3f, 0xff, 0xfc,
  0x7e, 0x00, 0x00, 0xff, 0xff, 0xfe, 0x1f, 0xff, 0xf0, 0x00, 0x7c, 0x00, 0x1f, 0xff, 0xf8,
};`

I put the image on the screen like this:
```
  u8g2.begin();
  delay(100);                                     // Wait for initialization process to finish
  u8g2.clearDisplay();                            // Clear display
  u8g2.clearBuffer();                             // clear the internal memory

  u8g2.drawXBM(0, 0, 120, 22, FESTO);         // Draw the logo
  u8g2.sendBuffer();                              // transfer internal memory to the display
  delay(1500);                                  // Wait 1,5s before the rest is put on the display
```

Am I doing something wrong or is this a bug?

BR,
Bart

olikraus, 2016-09-24:
strange, i guess i need to investigate this more closely.

olikraus, 2016-09-30:
Can you provide the original & correct bitmap, maybe as png.
Correct way to consturct this picture is to generate a xbm format, e.g with gimp..

olikraus, 2016-09-30:
I retested the bitmap procedure and it looks to me that this works.
changed label...

olikraus, 2016-10-01:
please create a new issue for further questions

closed 2016-10-01 by olikraus