olikraus/u8g2 #63: Issue with new SSD1322 driver

slackinfux, 2016-09-29:
I was attempting to get my SSD1322 display working on U8G2 yesterday (I have it working great on a Due with u8glib!) to no avail until you posted a code change for the display, yesterday. I was able to get the display partially working after your changes, but was having an issue where the screen was only drawing at 240x32, even though I uncommented the U8G2_16BIT line in u8g2.h.  I'm using the EastRising version of the display, in this case, although I do have an NHD version for testing as well.

I had a look at u8x8_d_ssd1322.c and after messing around with settings in the display info section, was finally able to get it to draw to the entire display. First issue was that the pixel_height was set to 32, not 64. 

The second issue seems to be that #ifdef U8G2_16BIT doesn't come out as true for some reason, so the tile width was getting set as 31 and the pixel_width to 256-16. I'm getting around that by just changing the values on the other side of the #else statement to 32 and 256 respectively, which works for me.

#ifdef U8G2_16BIT
  /* tile_width = */ 32,		/* 256 pixel, so we require 32 bytes for this */
#else
  /* tile_width = */ 32,		/* reduced to 240 pixel in 8 bit mode, looks odd, but at least something is visible */
#endif
  /* tile_hight = */ 8,
  /* default_x_offset = */ 0x01c,	/* this is the byte offset (there are two pixel per byte with 4 bit per pixel) */
  /* flipmode_x_offset = */ 0x01c,
#ifdef U8G2_16BIT
  /* pixel_width = */ 256,
#else
  /* pixel_width = */ 256,
#endif
  /* pixel_height = */ 64
};

olikraus, 2016-09-30:
Thanks. Both topics should be fixed now.

slackinfux, 2016-09-30:
That works.

closed 2016-09-30 by olikraus