olikraus/u8g2 #66: Add support for EA Dog / UC1611s displays

claudio755, 2016-10-03:
Hi Oliver, 
thank you so much for your great work! Are you still thinking about adding support for the great Electronic Assembly displays in your u8g2 library? I am using a  EA DOGM240 display with 240x64 px. As I am on an ESP8266 I unfortunately can not use the u8glib (It wont compile). 

Thanks so far
C

olikraus, 2016-10-04:
would you be willing to test the code?
If yes i could start to port this code to u8g2.

I also have a DOGM240 in my lab, but it's still not working :-(

claudio755, 2016-10-04:
Sure I'd support you whereever I can. Just let me know.

olikraus, 2016-11-13:
Current status: I have tested DOGXL240 with the I2C interface. This now works.
DOGM240: Not tested
SPI: Not tested (actually it does not work, but i do not know the root cause for this, maybe my test setup is wrong)
@claudio755: can you test dogm240 with SW spi? This would be this constructor:
```
U8G2_UC1611_EA_DOGM240_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/  8);
```
Of course, the pin numbers might be different for you.

You can download the library from 
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Installation should work through the Arduino IDE (add zip library if I remember correctly)

olikraus, 2016-11-13:
I have created another SPI test setup. SPI works as expected with the DOGXL240.

olikraus, 2016-11-13:
Everything works, both displays DOGM240 and DOGXL240 work correctly in I2C and SPI mode.
This issue can be closed.

closed 2016-11-13 by olikraus

claudio755, 2016-11-13:
Hi Oliver, 
I can confirm - it works! Thank you so much!