olikraus/u8g2 #68: Compiler warnings

jpoppe, 2016-10-06:
The amount of compiler warnings in u8g2 is already much less then in u8glib, but there are still a few left. I have been hesitating for a while before I opened this ticket since I don't want to bother you, but I don't know for sure if you are aware of them. I sometimes miss other ('important') warnings because of the noise. 

/home/jpoppe/Arduino/libraries/U8g2/src/U8x8lib.cpp:51:98: warning: unused parameter 'arg_ptr' [-Wunused-parameter]
 extern "C" uint8_t u8x8_gpio_and_delay_arduino(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
                                                                                                  ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_font.c: In function 'u8g2_font_calc_vref_font':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_font.c:927:46: warning: unused parameter 'u8g2' [-Wunused-parameter]
 u8g2_uint_t u8g2_font_calc_vref_font(u8g2_t *u8g2)
                                              ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_kerning.c: In function 'u8g2_GetKerning':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_kerning.c:47:33: warning: unused parameter 'u8g2' [-Wunused-parameter]
 uint8_t u8g2_GetKerning(u8g2_t *u8g2, u8g2_kerning_t *kerning, uint16_t e1, uint16_t e2)
                                 ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_kerning.c: In function 'u8g2_GetKerningByTable':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_kerning.c:78:40: warning: unused parameter 'u8g2' [-Wunused-parameter]
 uint8_t u8g2_GetKerningByTable(u8g2_t *u8g2, const uint16_t *kt, uint16_t e1, uint16_t e2)
                                        ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_polygon.c: In function 'pg_AddPolygonXY':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8g2_polygon.c:304:45: warning: unused parameter 'u8g2' [-Wunused-parameter]
 void pg_AddPolygonXY(pg_struct *pg, u8g2_t *u8g2, int16_t x, int16_t y)
                                             ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_8x8.c: In function 'u8x8_ascii_next':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_8x8.c:199:34: warning: unused parameter 'u8x8' [-Wunused-parameter]
 uint16_t u8x8_ascii_next(u8x8_t *u8x8, uint8_t b)
                                  ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_d_a2printer.c: In function 'u8x8_d_a2printer_common':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_d_a2printer.c:64:68: warning: unused parameter 'arg_int' [-Wunused-parameter]
 uint8_t u8x8_d_a2printer_common(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
                                                                    ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_d_ls013b7dh03.c: In function 'u8x8_d_ls013b7dh03_128x128':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_d_ls013b7dh03.c:70:71: warning: unused parameter 'arg_int' [-Wunused-parameter]
 uint8_t u8x8_d_ls013b7dh03_128x128(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
                                                                       ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_setup.c: In function 'u8x8_dummy_cb':
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_setup.c:39:31: warning: unused parameter 'u8x8' [-Wunused-parameter]
 uint8_t u8x8_dummy_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
                               ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_setup.c:39:45: warning: unused parameter 'msg' [-Wunused-parameter]
 uint8_t u8x8_dummy_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
                                             ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_setup.c:39:58: warning: unused parameter 'arg_int' [-Wunused-parameter]
 uint8_t u8x8_dummy_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
                                                          ^
/home/jpoppe/Arduino/libraries/U8g2/src/clib/u8x8_setup.c:39:73: warning: unused parameter 'arg_ptr' [-Wunused-parameter]
 uint8_t u8x8_dummy_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
                                                                         ^

olikraus, 2016-10-06:
fixed with commit 421889ca29eaf8f8dd694b3a21674402f785e484

closed 2016-10-06 by olikraus

jpoppe, 2016-10-06:
Awesome, thanks again!