olikraus/u8g2 #70: Microchip XC8 compiler issues

basilfx, 2016-10-08:
Some thing I encountered while trying to use the Microchip XC8 compiler:

* In `u8g2_fonts.c`, method `u8g2_SetFont`: the expression `if ( u8g2->font != font )` fails because the `font` property is `const` in `u8g2_t`. If I remove the check (or remove the const modifier), it succeeds. The warning I get is `u8g2_font.c:989: error: (712) can't generate code for this expression`
* It seems that the XC8 compiler determines that there is not enough memory before linking: `u8g2_d_memory.c:200: error: (1250) could not find space (3840 bytes) for variable F2253`. Similar errors occur for the other methods.
* Sorry for bringing up #6 again, but the XC8 compiler is terrible at parsing the font file: `u8x8_fonts.c:1429: warning: (228) illegal character (0x7F)`. There are many of these errors. I tried to modify `bdfconv` to output as integer arrays (XC8 is reasonably fast at parsing that), but then it fails because it cannot generate code for a 12MB file (I get malloc errors on my Mac). `USE_LARGE_FONTS` is disabled.
* There are several other warnings that I think can be ignored: for instance it throws warnings for message callbacks that are invoked with NULL as (last) argument.

Most problems can be worked around, e.g. removing the setup methods and fonts that you don't need.

Regarding the fonts, maybe it is a good idea to readd the `fontsrc/` folder, so one can decide to use the `u8g2_fonts.c` file, or select the fonts desired?

olikraus, 2016-10-08:
If expression: I definitly do not know what is wrong here. From my perspective this is all ok.

olikraus, 2016-10-09:
Other problems: To me the gcc had been a reference compiler for a long time and i have full trust to what gcc tells me. I had worked in the past with XC8. XC8 gut cut down a lot by microchip. Actually i am happy that gcc works with almost no warnings.
Other aspect is this: u8g2 has been optimized for Arduino AVR + ARM toolchains. This means there is a single font file with the strings containing the font data (for speed optimization)

I also do not understand the warning "warning: (228) illegal character (0x7F)" at all.

basilfx, 2016-10-09:
> To me the gcc had been a reference compiler for a long time and i have full trust to what gcc tells me. I had worked in the past with XC8. XC8 gut cut down a lot by microchip. Actually i am happy that gcc works with almost no warnings.

I fully understand and I definitely think that's the way to go. The things I noticed are easy to work around.

> Other aspect is this: u8g2 has been optimized for Arduino AVR + ARM toolchains. This means there is a single font file with the strings containing the font data (for speed optimization)

I understand, but having both `csrc/u8g2_fonts.c` and `fntsrc/<font_name>.c` would help :-) That would minimize editing the source files, which complicates update process.

> I also do not understand the warning "warning: (228) illegal character (0x7F)" at all.

Me neither. User manual says `This character is illegal in the C code. Valid characters are the letters, digits and those comprising the acceptable operators.`, but as far as I can tell, the source file consists of valid ASCII characters. Guess it's a bug in the compiler then?

olikraus, 2016-10-09:
> I understand, but having both csrc/u8g2_fonts.c and fntsrc/<font_name>.c would help :-) That would minimize editing the source files, which complicates update process.

of course, this could be done

basilfx, 2016-10-29:
> Will you create an issue?

Done, see #78 (I forgot to create the issue)

So far I was able to get U8g2 running using the XC8 compiler, but it required editing many files. For some troubles, we have Microchip looking into them. Will report back.

olikraus, 2016-10-29:
I have asked Microchip already. There is nothing what will happen. Microchip has removed a simple optimization feature from the free version of their compiler.

basilfx, 2016-10-29:
We have a license on the full version, and we are in contact with their support team. They are active looking into the remaining issues.

olikraus, 2016-10-29:
wow...

olikraus, 2017-01-04:
closing this to keep the number of open issues limited. This topic can be reopend of course...

closed 2017-01-04 by olikraus

prabhat73362, 2021-02-13:
../../../../Desktop/u8g2-master/csrc/u8g2_d_memory.c:595:: error: (1250) could not find space (12000 bytes) for variable F20947

../../../../Desktop/u8g2-master/csrc/u8g2_d_memory.c:1024:: error: (1250) could not find space (9600 bytes) for variable F21064

../../../../Desktop/u8g2-master/csrc/u8g2_d_memory.c:1519:: error: (1250) could not find space (11520 bytes) for variable F21199

how can I remove these warnings please help me I am using Xc8 free version compiler

olikraus, 2021-02-14:
Noone can help. You must buy the professional version of xc8