olikraus/u8g2 #74: Add support for println

camp0s, 2016-10-12:
Maybe I've missed it, but it would be handy to have a print and newline command integrated.

olikraus, 2016-10-12:
What would be the precise action?  Set x to 0 and advanced y by current line height?

camp0s, 2016-10-13:
Yes, that would be the objektive. I tryed u8x8.print("Text\n"); but it didn't recognize the new line char.

I wrote a small function in my program: I call it after every `.print`, it will delete the next row and ready to print a new line.
```
int row=0; // global var

....

void nextLine() {
  row++;
  if (row>=8) row=0;
  u8x8.drawString(0,row,"                ");
  u8x8.setCursor(0,row);
}
```

olikraus, 2016-11-04:
Arduino implementation:
size_t Print::println(void)
{
  return write("\r\n");
}

olikraus, 2016-11-04:
Implemented for u8x8.
Some testing required.

olikraus, 2016-11-05:
Added test code to U8x8 Graphics test --> All is fine

closed 2016-11-05 by olikraus

keith204, 2016-11-16:
Are there plans to do the same for u8g2?  I was happy to use this in u8x8 a couple of days ago, and then switched to u8g2 and noticed it wasn't there.  This would be super handy.

olikraus, 2016-11-16:
it will occupy more flash-ROM...

keith204, 2016-11-16:
> it will occupy more flash-ROM...
personally, I'm very much OK with that.  But, if you feel it's best to leave it out for that reason, that would make sense.

olikraus, 2016-11-17:
Well, yes that was indeed my reason not to implement this for u8g2. It is simple enough for u8x8, but for u8g2 more statements are required.