olikraus/u8g2 #77: Where do I define the I2C addr of OLED?

k5map, 2016-10-19:
I looked through the examples and do not see where I define the 0x3D address of my OLED display.  Can someone educate me on where this is defined?

olikraus, 2016-10-19:
This function is not (yet) there on the cpp interface. As workaround you could use this:
u8g2_SetI2CAddress(u8g2.getU8g2(), 0x3d);

If i remember for SSD1306, address defaults to 0x03c.

Todo: Add this function to cpp wrapper.

olikraus, 2016-11-04:
implemented and wiki updated

closed 2016-11-04 by olikraus

mtfurlan, 2019-10-14:
As I found this on google first, here is the link for the next person:
https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

temurchichua, 2023-06-27:
> Update: The workaround is wrong. It should be: u8g2_SetI2CAddress(u8g2.getU8g2(), 0x3d*2);

Hey, I'm using the same module and I2C scan told me that the display device is at 0x3C but I still had to write `u8g2.setI2CAddress(0x3C * 2);`.

Where is that `* 2` coming from?

olikraus, 2023-06-27:
https://github.com/olikraus/u8g2/issues/821