olikraus/u8g2 #78: Provide csrc/u8g2_fonts and fntsrc/*.c

basilfx, 2016-10-29:
For compilers that are less capable of sorting out which files to include, provide both the fonts combined and separated. 

For instance, the XC8 compiler has troubles with so many fonts.

Relate to #70

olikraus, 2016-10-29:
The fonts are only one of several problems. Also if i would store each font in a single file, there are other things which will break XC8. I already tried hard to port u8glib to XC8, but finally i gave up. In fact i do not want to waste time on XC8. gcc is much better and for free.

olikraus, 2016-11-02:
the font build process will be updated, so that the single font files are stored in
u8g2/tools/font/build/single_font_files

olikraus, 2016-11-02:
font files are now available here:

https://github.com/olikraus/u8g2/tree/master/tools/font/build/single_font_files

Hope this is good enough.

closed 2016-11-02 by olikraus

basilfx, 2016-11-02:
Yes, thank you very much. This works much better!