olikraus/u8g2 #79: I2C Fast mode info request

qrpfun, 2016-10-31:
In u8glib there was a FAST mode for I2C mode on 1306.
Is this mode enabled by default or is not actually implemented ?
This mode should be available in all 3 modes, even in u8x8 text only lib ?
tnx in advance for your support

olikraus, 2016-11-02:
Good point. I think this is missing in procedure u8x8_byte_arduino_hw_i2c (U8x8lib.cpp)

ToDo:
In U8X8_MSG_BYTE_INIT or U8X8_MSG_BYTE_START_TRANSFER, set the transfer speed with 
if (  u8x8->display_info->i2c_bus_clock_100kHz >= 4 )
   Wire.setClock(400000);

 * [ ] Implementation
 * [ ] Test

olikraus, 2016-11-02:
setClock is there only since recent versions. It is at least not there with 1.0.x and 1.5.0. Maybe it got introduced with 1.6.x

olikraus, 2016-11-02:
tested with Adafruit Featherwing OLED. Speed improvement by 50% .. 100%

closed 2016-11-02 by olikraus

MarsLoop, 2017-03-23:
Could you please add a constructor for Fast I2C in combination with an SSD1306 without a reset line?

I'm using the following constructor, but its pretty slow: 

U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // OLEDs without Reset of the Display