olikraus/u8g2 #80: No T6963 240X64 mode

RndMnkIII, 2016-11-01:
Hello, can you add support for T6963 240X64 screen mode.

olikraus, 2016-11-02:
I need a little support for this: How does the current 240x128 appear? Does it work except for the fact that the screen is too large? Can you sent a picture of the u8g2 logo?

RndMnkIII, 2016-11-02:
I ve tested 240x128 limiting to 240x64 working ok
Output from FPS program:
https://drive.google.com/file/d/0By5wADoEOJ9fOG5mZ1hzbGQzdjA/view?usp=sharing
https://drive.google.com/file/d/0By5wADoEOJ9fS3NsSVNScW8yakE/view?usp=sharing
https://drive.google.com/file/d/0By5wADoEOJ9fMEFGOVlVM0VUck0/view?usp=sharing
https://drive.google.com/file/d/0By5wADoEOJ9fdDJJZGxMejF1WHM/view?usp=sharing

RndMnkIII, 2016-11-02:
I've using a teensy 3.2 with logic levels converters from 3.3v to 5v. I'm not sure if my T6963 can accept 3.3v signals

RndMnkIII, 2016-11-02:
Seems not particular fast, The T6963 have the culprit?

olikraus, 2016-11-02:
Thanks for the pictures. I will create a 240x64 device based on this.
> Seems not particular fast, The T6963 is the culprit?
Speed will double with the 240x64 display. In general the parallel interface is not so much optimized to keep Arduino compatibility.

RndMnkIII, 2016-11-02:
Can you help me, please? I've have to choose between 3.2" 256X64 oled display  with SSD1322 controller or  a %"  LCD t6963c 240x64. Which controller is fastest? The SPI interface can be faster than parallel? The Teensy 3.x SPI vs Arduino SPI can be faster? The Teensy 3.x overclocking can made faster, with SPI or Parallel?
Thanks!

christophepersoz, 2016-11-02:
Hello, SPI on Teensy is much faster than the Arduino's one - 5MHz at maximum speed. I made some test with a SSD1322 on Teensy 3.2 board, here the link of the results : https://forum.pjrc.com/threads/23445-Teensy-3-0-and-u8glib?p=86456&viewfull=1#post86456

Hope that can helps.

RndMnkIII, 2016-11-02:
Thank you. I'll try the SDD1322 in SPI mode when I receive the order

RndMnkIII, 2016-11-02:
For testing purposesI've tested the T6963 with U8GLIB_T6963_240X64  and U8GLIB_T6963_240X128 with not success.

olikraus, 2016-11-02:
I think U8glib will not work with Teensy.

christophepersoz, 2016-11-02:
I does, but you need to use a wrapper, and it works quite fast :D

RndMnkIII, 2016-11-02:
the U8GLIB works with the KS0108, tested with 192x64 screen. Can be something related to the controller initialization?

olikraus, 2016-11-02:
The KS0108 parallel interface differs from the T6963 interface. So it might be caused by that interface. 
But there might be also an issue with the wrapper or the controller itself. One of the goals of u8g2 had been a more portable low level interface. At least for u8g2 I hope it will work more reliable with the Teensy.

RndMnkIII, 2016-11-02:
Tested without logic level converters. My T6963 works OK with 3.3v signaling, remaining 5v for VDD, LED_A.

RndMnkIII, 2016-11-02:
In u8g2 where actually the data is written to data pins?. I want to replace the generic digitalWriteFast by byte write to GPIOD_PDOR where actually is mapped my teensy 3.2 data pins.

olikraus, 2016-11-02:
You have to create a new byte output procedure:
u8x8_byte_8bit_6800mode
or
u8x8_byte_8bit_8080mode
both are defined in u8x8_byte.c
The byte procedure is part of the constructor in U8g2lib.h

RndMnkIII, 2016-11-02:
In `uint8_t u8x8_byte_8bit_8080mode`, changed the following:
```
	//Added by RndMnkIII
	//GPIOD_PDOR = GPIOD_PDOR | (*data);
	GPIOD_PDOR = b;
	data++;
	arg_int--;
	// Commented out by RndMnkIII
	// for( i = U8X8_MSG_GPIO_D0; i <= U8X8_MSG_GPIO_D7; i++ )
	// {
	  // u8x8_gpio_call(u8x8, i, b&1);
	  // b >>= 1;
	// }  
```
Resulting in the following test results:
-draw clip test 62.5
-clear screen 51.9
-draw @ 22.9
-draw pixel 28.2

See the scrolling text is more smooth, using 1 pixel step:
https://drive.google.com/file/d/0By5wADoEOJ9fcmZNVzE4WnZrUnc/view?usp=sharing

RndMnkIII, 2016-11-02:
As side effect I've to use shorter cables because I've started to observe pixel artifacts in the screen. It's not too visible. I dont't know that this is related to my modifications or the greater speed at the microcontroller writes to the screen.

olikraus, 2016-11-02:
Excellent scrolling :-)

Obviously, there is always a tradeoff between speed, code clarity and portability. If it had been easy enough for you to increase speed for your specific environment and setup, then I am happy.

olikraus, 2016-11-02:
I have added the new constructor for the T6963 240x64 display. It is included here for download
https://github.com/olikraus/U8g2_Arduino

RndMnkIII, 2016-11-02:
Thank You. Tested and working!

RndMnkIII, 2016-11-02:
Adding delayMicroseconds() of 4-5  after GPIOD_PDOR = b; almost remove pixel artifacts reducing a bit the performance but improving the appearance.

olikraus, 2016-11-02:
yes, i guess there must be a minimal delay

olikraus, 2016-11-02:
can we close this issue?

closed 2016-11-02 by RndMnkIII