olikraus/u8g2 #82: Font character memory size limit doubt?

RndMnkIII, 2016-11-02:
In u8glib there was a limit in character size of 256 bytes when converting from bdf font. This limit exist in u8g2, can have a character bigger in memory size?

olikraus, 2016-11-02:
The same limit exists, however the data is now compressed, so that the actual size can be much bigger.  The limit of 256 bytes is true for the compressed glyph data.
Which size do you need?
Some of the fonts are already very big, see here: https://github.com/olikraus/u8g2/wiki/fntlistall

RndMnkIII, 2016-11-02:
I'm using FontForge to generate the bdf. I'll try the u8g2 converter with the same font type and size.

RndMnkIII, 2016-11-02:
Tested with 
[Hiruko-BlackAlternate-56.zip](https://github.com/olikraus/u8g2/files/566483/Hiruko-BlackAlternate-56.zip)
bdfconv -b0 -f1 -oHiruko56.h -nHiruko56 Hiruko-BlackAlternate-56.bdf

Converted without any warning.

https://drive.google.com/file/d/0By5wADoEOJ9fM1lKcXR0ME5fdnM/view?usp=sharing
[image]

olikraus, 2016-11-02:
Wow, very nice. Good work.
The pixel errors are optical issues? Is this a picture of a scrolling text?

RndMnkIII, 2016-11-02:
The blurring it's because the scrolling text, but the blurring it's better with this screen that with a ks0108 tested from the same manufacturer (BuyDisplay.com)

RndMnkIII, 2016-11-02:
Scrolling with 3 pixels steps

RndMnkIII, 2016-11-02:
Video for better idea of results:
https://drive.google.com/file/d/0By5wADoEOJ9fRnRiT0o2bm12UWc/view?usp=sharing

olikraus, 2016-11-02:
Cool. Nice effect.
Thanks for sharing.
So, this issue can be closed?

closed 2016-11-02 by RndMnkIII