olikraus/u8g2 #87: ST7920 in SPI Mode is very slow

floyd871, 2016-11-16:
I want to switch to the new v2 version. But the Display with the v2 ist really VERY slow. Up to 4-5 TIMES.
My setup Arduino Nano with ST7920 Display SPI.
I wrote this code to switch between the two version.  I don't want to use the "F" Contructor, because this takes too much ram.
What can i do?

```
//#define V1

#ifdef V1
  #include <U8glib.h>
  U8GLIB_ST7920_128X64_1X     u8g(/* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 8);
#else
  #include <U8g2lib.h>
  U8G2_ST7920_128X64_1_SW_SPI u8g(U8G2_R0,/* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 8);
#endif

int yPos = 0;

void setup(){
  #ifndef V1
    u8g.begin();  
  #endif    
  u8g.setFont(u8g_font_unifont);
  u8g.setColorIndex(1); // Instructs the display to draw with a pixel on. 
}

void loop() {  
  u8g.firstPage();
  do {  
    draw();
  } while( u8g.nextPage() );
  
  // If its too fast, you could add a delay
  if(yPos < 83){
    // if it's too slow, you could increment y by a greater number
    yPos++;  }
  else{
    // When the yPos is off the screen, reset to 0.
    yPos = 0;
  }  
}  

void draw(){
  u8g.drawStr( 0, yPos, "Hello World");    
}

```

olikraus, 2016-11-16:
HW SPI should be much faster.
SW SPI is not optimized for speed.

floyd871, 2016-11-16:
Yes with HW SPI it is fast as v1. I have used the SW SPI constructor because the params were similar to the v1 constructor. I had already connected the Display to the HW pins, so the Constructor with only one parameter (CS) had confused me. Thanks for your fast help.

olikraus, 2016-11-16:
Yes, good point.

olikraus, 2016-11-26:
I found an issue, which as casued the low speed. Now SPI speed is 10 times faster.

closed 2016-11-26 by olikraus