olikraus/u8g2 #88: Can add support for KS0108

RndMnkIII, 2016-11-19:
Hello again Olikraus, can you add suport in u8g2 for the KS0108 controller?
Thanks!

olikraus, 2016-11-19:
OpenGLCD is probably faster. You still want u8g2 support for this display?

RndMnkIII, 2016-11-19:
Yes because I need:
- Proper screen clipping
- Big font characters
- Extensive UTF-8 support

With OpenGLCD I do not have support or I am not totally satisfied with the results obtained.

RndMnkIII, 2016-11-19:
The speed or memory ussage is not an issue with the Teensy 3.2

olikraus, 2016-12-03:
Introduce a new transfer start and end procedure with bit pattern for the CS signals as arguments.

Create a new byte transfer function which uses this new transfer start & end.

olikraus, 2016-12-03:
Ks0108: looks like 6800 mode. R/W must be low. E latches with falling edge.

olikraus, 2016-12-03:
cad001 or cad011 (no arts used)

olikraus, 2016-12-06:
@RndMnkIII, you oncementioned a 192x64 controller, right? What are the cs0, cs1, cs2 settings for this display? Any datasheet?

olikraus, 2016-12-06:
Todo: Optimization

RndMnkIII, 2016-12-06:
http://www.buydisplay.com/default/4-inch-192x64-lcd-module-graphic-display-ks0107-ks0108-black-on-white
There are schematics, datasheet and source code example (note the chip selection is active to low).

olikraus, 2016-12-07:
ERM19264
left: 011, middle: 101, right: 110, no chip select: 111

RndMnkIII, 2016-12-07:
Yes, That's what I did  for implement it on u8glib.

olikraus, 2016-12-08:
ok, i have added both devices to 2.8.2 from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Maybe you can check the 192x64 GLCD and give me some feedback

Still not improved for speed on ATMega architectures...

olikraus, 2016-12-11:
ok, i will close this... 
If you have any feedback on the 192x64 GLCD i would be glad to hear about that.

closed 2016-12-11 by olikraus