olikraus/u8g2 #89: Support for SSD1306 64x48?

cedbeu, 2016-11-20:
Hi, am using the WeMos OLED Screen that is 64x48px using the SSD1306 driver/controller. I can initialize it with U8G2_SSD1306_128X64_NONAME_F_SW_I2C (clock: 5, data: 4, reset: 13) but I cannot figure out how to modify the driver to set the display size to 64x48. Seems this size was supported in u8glib…?

olikraus, 2016-11-20:
Should be possible...
 * [ ] locate 0.66 in my lab (order date Feb 07, 2015)
 * [ ] create device for the 64x48 device 
 * [ ] testing...

Notes: According to u8glib visible area is horizontal centered and starts at 32 (0x010|2, upper 4 bit set to 2).
Vertical visible area seems to start a 0.

olikraus, 2016-11-20:
implemented & tested.
Download prerelease (2.7.0) here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

closed 2016-11-20 by olikraus

cedbeu, 2016-11-21:
Wow! Great, thanks, works perfect.