olikraus/u8g2 #90: ST7920 parallel mode (8080 / 6800) not working

viti95, 2016-11-20:
I've made a straight port of a benchmark I wrote with U8Glib, and while it works perfectly in HW / SW SPI modes, it doesn't in any of the two parallel modes. I've tested all the possible combinations on the constructor (enable, cs, ds) to see if it was my mistake, but none of them work (I found the description of the parameters in the new constructor really confusing). These are the initializations i'm using:

U8GLIB_ST7920_128X64_4X u8g(52, 51, 10); -> Working
U8GLIB_ST7920_128X64_4X u8g(10); -> Working
U8GLIB_ST7920_128X64_4X u8g(37, 35, 33, 31, 29, 27, 25, 23, 52, 10, 51); -> Working

U8G2_ST7920_128X64_2_SW_SPI u8g2(U8G2_R2, 52, 51, 10); -> Working
U8G2_ST7920_128X64_2_HW_SPI u8g2(U8G2_R2, 10); -> Working
U8G2_ST7920_128X64_2_8080 u8g2(U8G2_R2, 37, 35, 33, 31, 29, 27, 25, 23, 52, 10, 51); -> NOT working
U8G2_ST7920_128X64_2_6800 u8g2(U8G2_R2, 37, 35, 33, 31, 29, 27, 25, 23, 52, 10, 51); -> NOT working

The U8G2 version is 2.6.3, and the device is an Arduino Mega 2560

PS. Please forgive my English

olikraus, 2016-11-22:
hmmm... i have to check this....

olikraus, 2016-11-25:
it is working, however you have to set the R/W input of the display to 0. Additionally there is no CS in parallel mode of the ST7920, so in the constructor CS has to be set to U8X8_PIN_NONE.

For me, this works fine:

U8G2_ST7920_128X64_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 18 /* A4 */, /*cs=*/ U8X8_PIN_NONE, /*dc/rs=*/ 17 /* A3 */, /*reset=*/ 15 /* A1 */);	// Set R/W to 0,

closed 2016-11-26 by olikraus

olikraus, 2016-11-26:
I added some description here:
https://github.com/olikraus/u8g2/wiki/gallery#26-nov-2016-st7920-128x64-lcd-in-8080-parallel-mode

viti95, 2016-11-26:
Tested and working!!!! Thanks for the great work!!!!
[image]

olikraus, 2016-11-26:
Very nice.  Thanks for sharing.

kivikakk, 2018-01-01:
Thank you so much for this thread! It helped me get my ST7920 working in parallel mode as well, which is so much faster than SPI. 😊

kivikakk, 2018-01-01:
Note to anyone else following: don't just change `u8x8_byte_4wire_sw_spi` to `u8x8_byte_8bit_8080mode`, you need to change `u8g2_Setup_st7920_s_128x64_f` to `u8g2_Setup_st792_p_128x64_f` too! I was reading `u8x8_byte.c` wondering how just swapping one function out could cause it to operate in parallel mode, and apparently I was right for wondering, there was definitely more to it than that!

olikraus, 2018-01-01:
Correct