olikraus/u8g2 #91: UC1698 compatibility

pierbiag, 2016-11-21:
Hello, I was planning to buy a ERC240128FS-1 with the UC1698 controller.
However I was set back by the fact that (apparently) there are no libraries available and i wouldn't know where to start.
Are you planning to include support for that controller in the near (1-2 months) future?
Thank you in advance,
Pier

olikraus, 2016-11-22:
There are no plans to add support for this display. I once had considered to do so, but there are two points which lead to the decision not to implement support for this controller:
1. The UC1698 is an RGB controller used for a BW display. It is very complicated to use and somehow.
2. According to my own discussion with EastRising/BuyDisplay i had the impression that they wanted to move to a different controller.

pierbiag, 2016-11-22:
Thank you very much for the quick reply! I will try with something else!

Pier Maria Biagiolini

olikraus, 2016-11-22:
There is already support for the UC1611 and maybe i will add support for the UC1608 also.

olikraus, 2016-11-23:
uc1608 is now supported

closed 2016-11-23 by olikraus

pierbiag, 2016-11-23:
Thank you very much for your help again!
In the end I will go with an EA DOGL128, assuming that i can use the same constructor of the DOGM128 (correct me if I'm wrong).

olikraus, 2016-11-23:
Yes, to my knowledge dogm128 and dogl128 are hardware compatible.

pierbiag, 2016-11-23:
perfect!

nicgon, 2020-08-03:
Hi Friends!.
I currently have a JHDLCD 240X128 Driver uc1698u display.
I'm working in it.
I have found code files to make it work but I am not an expert programmer.
I have always used armed libraries for arduino.
Can I share these files with you? I have direct contact with JHDLCD Factory.

Thanks for you work!

Adesin-fr, 2023-03-21:
Hi,

@nicgon , did you manage to get the uc1698 working ? 
I also bought displays with this controller !

Regards

olikraus, 2023-03-21:
From what I remember: It is almost impossible to add support for this display driver. At least the effort would be huge. I personally would suggest to look for different displays. If you want to continue with this LCDs, please create a new issue or start a discussion here...