olikraus/u8g2 #93: Improvement Request: multiple display buffer areas and memcpy to display memory

RndMnkIII, 2016-11-23:
Hi, I've thinking that can be very useful to define procedures for graphic memory area  ucontroller to draw offscreen graphics and for  graphics animated fx and to copy this area to display controller memory.

RndMnkIII, 2016-11-23:
Better explained: to have possibility to reserve memory for draw on it and transfer this memory to display controller memory without using an intermediate buffer

olikraus, 2016-11-23:
This function already exists, but is not documented and not available on the Arduino level.

uint8_t u8x8_DrawTile(u8x8_t *u8x8, uint8_t x, uint8_t y, uint8_t cnt, uint8_t *tile_ptr);

This function has some drawbacks. Mostly it depends on the memory architecture of the target display. This means a bitmap transferred by this function will work on one display but will be garbled on others.

Well.. I guess this is what you want if you want to bypass u8g2 bitmap procedures, right?

RndMnkIII, 2016-11-23:
I had thought that for example when you want to print text with u8g2, pass that method an extra field with a pointer to the area of ​​memory outside the screen and render it there using XOR,AND,... blending as an option. Another idea is render directly to the tilebuffer for create graphics fx: rotozoomer text, plasma animation, etc and memory copy to display controller memory.

RndMnkIII, 2016-11-23:
The teensys 3.5 and 3.6 with fpu unit, 1xxMhz speed and 128kb RAM  are very capable of this.

RndMnkIII, 2016-11-23:
The bitmaps can be preprocessed for the target display memory organization to speed up drawing

RndMnkIII, 2016-11-23:
Many years ago I remember the existence of compiled bitmaps converted directly to machine code for speed up drawing on slow intel cpu 386. This is too xtreme and cpu architecture sensitive (needs small instructions size for to be faster than memcpy) but the idea is adjust the bitmap byte ordering to the screen memory organization to avoid checks on runtime

olikraus, 2016-11-23:
Well, yes, I kept the memory architecture identical to the memory architecture of the target controller. There is only one graphics primitive which maps to the target memory architecture: a line drawing procedure for vertical and horizontal lines. All u8g2 price just call this one function.
Thinking about XOR, it would be simple to implement this, however it might slow down the project a lot if we add more functionality into this procedure.

olikraus, 2016-11-23:
One more idea: I could provide a hook function,  which allows to pre fill the u8g2 buffer with some kind of background image. After that, the normal u8g2 can draw over it.

RndMnkIII, 2016-11-23:
Can be create alternate method using C++ overload and use the method needed, the direct rendered or the offscreen rendered

RndMnkIII, 2016-11-23:
Yes the hook function seems good idea and left the user do all the low level drawing stuff needed and blend it if needed with u8g2 primitives. For example a background plasma with foreground title text or scrolling text

olikraus, 2016-11-23:
Okay, then let me do the following:
 * [ ] Provide and document the draw tile procedure
 * [ ] Implement a hook/callback function for the u8g2 buffer

olikraus, 2016-11-23:
Will this help?

olikraus, 2016-11-24:
Added documentation to u8x8 interface.

RndMnkIII, 2016-11-26:
I'll try drawTile for testing memory organization and transfer performance witn teensies Auduino Mega and teensies 3.2 and 3.5

olikraus, 2016-11-26:
```
void u8g2_SetBufferCurrTileRow(u8g2_t *u8g2, uint8_t row) U8G2_NOINLINE;

#define u8g2_GetBufferPtr(u8g2) ((u8g2)->u8g2->tile_buf_ptr)
#define u8g2_GetBufferTileHeight(u8g2)	((u8g2)->->tile_buf_height)
#define u8g2_GetBufferTileWidth(u8g2)	(u8g2_GetU8x8(u8g2)->display_info->tile_width)
/* the following variable is only valid after calling u8g2_FirstPage */
#define u8g2_GetBufferCurrTileRow(u8g2) ((u8g2)->tile_curr_row)
```

instead of the hok function, i guess i will better provide a suitable api

RndMnkIII, 2016-11-26:
The text printing  after tile buffer manipulation preserves backgroung (masking glyph shape against background image)?

olikraus, 2016-11-26:
I do not know exactly what you mean, but i plan to create an example which creates a custom background.

olikraus, 2016-11-26:
code will look like this. Instead of the hook function, i just copy something to the buffer before doing the normal draw functions.
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/DirectAccess/DirectAccess.ino

olikraus, 2016-11-26:
documentation required

olikraus, 2016-11-26:
testrelease is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

closed 2016-11-27 by RndMnkIII