olikraus/u8g2 #94: Required display support for Marlin

olikraus, 2016-11-25:
Marlin has support for the following displays:
DOGLCD (ST7565R)
REPRAPWORLD_GRAPHICAL_LCD (ST7920)
VIKI2 / miniVIKI (C12864)
ELB_FULL_GRAPHIC_CONTROLLER (ST7565 / LM6059_AF)
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (ST7920)
MAKRPANEL (C12864)
MINIPANEL (MINI12864)
CARTESIO_UI (DOGM128)
SAV_3DGLCD (SSD1306 / SH1106)

…or to put it another way…

U8GLIB_ST7920_128X64_4X
U8GLIB_ST7920_128X64_RRD
U8GLIB_DOGM128
U8GLIB_LM6059
U8GLIB_NHD_C12864
U8GLIB_SSD1306_128X64
U8GLIB_SH1106_128X64
U8GLIB_MINI12864

For U8g2 situation is like this:

SW SPI needs to be analyzed. paralle interface can be improved

U8GLIB_ST7920_128X64_4X --> Ported
U8GLIB_ST7920_128X64_RRD --> Not sure what this is
U8GLIB_DOGM128 --> Ported
U8GLIB_LM6059 --> Controller is ported, but the specific display setup needs to be analysed
U8GLIB_NHD_C12864 --> Controller is ported, but the specific display setup needs to be analysed
U8GLIB_SSD1306_128X64 --> Ported
U8GLIB_SH1106_128X64 --> Ported
U8GLIB_MINI12864 --> Controller is ported, but the specific display setup needs to be analysed

see u8glib#447

landodragon141, 2016-11-25:
> U8GLIB_ST7920_128X64_RRD --> Not sure what this is
RRD = RepRapDiscount

olikraus, 2016-11-25:
Ok, but I assume for u8glib, this is Just a st7920 Controller, right?

Blue-Marlin, 2016-11-25:
> Just a st7920 Controller, right?

Right. U8GLIB_ST7920_128X64_RRD is a software SPI version of a u8glib device for a ST7920 as it can be found on a http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller, but in priniple works on all ST7920.
It was made to make use from 'fastio' and was, i guess, one of the first '4x' capable u8glib devices around. Now speed is tweakable, in some aspects, until ST7920 hardware limits are exceeded. Even hardware SPI could not be faster in this aspects.

olikraus, 2016-11-25:
Cool. Thanks for the feedback.

olikraus, 2016-12-17:
I think this is almost done from my end. Let me know if there is any display which does not fit.
Assigned to release 2.9 which includes the zolen display.

closed 2016-12-17 by olikraus