olikraus/u8g2 #97: 4-bit support

jcaywood, 2016-12-01:
Would you consider adding 4-bit communications support for the ST7920 displays, it looks like only 8-bits is supported. Some of the 3D printer controller displays (with integrated rotary encoder and speaker) seem to use this display in that mode using the 4 high bits of the the display controller, along with the Enable and RS lines and it would be great to integrate these displays into other projects. Thanks for your consideration.

olikraus, 2016-12-01:
I can have a look at this. However i wonder how this was handled in history. The old u8glib did not have 4-bit support.

AnHardt, 2016-12-01:
@jcaywood 
Do you have an example model (type)?
At least for Marlin i know - all supported ST7920 displays are ether software are or hardware SPI. Do we miss one?

jcaywood, 2016-12-01:
Hello, after reading AnHardt's response I thought I must be missing something. I went through and did more research. It is indeed SPI. The listed pinouts for the RepRapDiscount display have E, RS and 4 bits defined for the display(LCD_RS, LCD_E, LCD4, LCD5, LCD6 and LCD7) which led me to believe it was running in 4-bit mode.  In reality only 3 of those lines (RS, E and LCD4) are connected through as I found on a schematic capture for the RRD display.  LCD5, LCD6, LCD7 are unused. I was able to use the following constructor to communicate with the display.  

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 23 /* A4 */ , /* data=*/ 17 /* A2 */, /* CS=*/ 16 /* A3 */, /* reset=*/ U8X8_PIN_NONE);

I apologize for taking up your time and thank you for your responses.

olikraus, 2016-12-01:
ok, so i can close this?

jcaywood, 2016-12-01:
Yes, please do. Thank you for your work on this library, it's very much
appreciated.

olikraus, 2016-12-01:
ok, great. Will keep 4 bit support in my mind. Maybe once i have time to implement this.

closed 2016-12-01 by olikraus