olikraus/u8g2 #98: SSD1306: Use precharge to increase the range for the brightness control

olikraus, 2016-12-02:
> I'm using it with a SSD1306 OLED. Unfortunately the hardware dim function as only
> minimal effect on brightness. I found out that the Pre--charge command can dim the
> display over a wider range (tested with the adafruit lib).
> How can I include a setPreCharge command in the u8g2 lib?

p-tric, 2016-12-02:
I found out to add:
in u8x8.h:
Line 406: #define U8X8_MSG_DISPLAY_SET_PRECHARGE 114
...
void u8x8_SetContrast(u8x8_t *u8x8, uint8_t value);
void u8x8_SetPrecharge(u8x8_t *u8x8, uint8_t value);

in u8x8_d_ssd1306_128x64_noname.c:
    case U8X8_MSG_DISPLAY_SET_PRECHARGE:
      u8x8_cad_StartTransfer(u8x8);
      u8x8_cad_SendCmd(u8x8, 0x0D9 );   /* ssd1306 PreCharge Command */
      u8x8_cad_SendArg(u8x8, arg_int );	/* ssd1306 has range from 0 to 255 */
      u8x8_cad_EndTransfer(u8x8);
      break;

in  u8x8_display.c:
void u8x8_SetPrecharge(u8x8_t *u8x8, uint8_t value)
{
  u8x8->display_cb(u8x8, U8X8_MSG_DISPLAY_SET_PRECHARGE, value, NULL);  
}

in  u8g2.h:
#define u8g2_SetPrecharge(u8g2, value) u8x8_SetPrecharge(u8g2_GetU8x8(u8g2), (value))

in   U8g2lib.h:
    void setPrecharge(uint8_t value) {
      u8g2_SetPrecharge(&u8g2, value); }

in  Arduino *.ino:
    u8g2.SetPrecharge(0x10);
enables fulll brightness control with
   u8g2.SetContrast(0 ... 255)

cu

olikraus, 2016-12-02:
wow, this already looks like a full solution. You nicely went deep into the inner structures of u8g2. Cool.

olikraus, 2016-12-03:
according to my tests, the precharge has only little influence. However the vcom deselect should be set to 0.

olikraus, 2016-12-03:
I did a lot of testing. Acutally i think we just need a different value for the vcom deselect level. It should be zero.

```
// value from 0 to 7, higher values more brighter
void setSSD1306VcomDeselect(uint8_t v)
{	
  u8x8_cad_StartTransfer(u8g2.getU8x8());
  u8x8_cad_SendCmd(u8g2.getU8x8(), 0x0db);
  u8x8_cad_SendArg(u8g2.getU8x8(), v << 4);
  u8x8_cad_EndTransfer(u8g2.getU8x8());
}

// p1: 1..15, higher values, more darker, however almost no difference with 3 or more
// p2: 1..15, higher values, more brighter
void setSSD1306PreChargePeriod(uint8_t p1, uint8_t p2)
{	
  u8x8_cad_StartTransfer(u8g2.getU8x8());
  u8x8_cad_SendCmd(u8g2.getU8x8(), 0x0d9);
  u8x8_cad_SendArg(u8g2.getU8x8(), (p2 << 4) | p1 );
  u8x8_cad_EndTransfer(u8g2.getU8x8());
}

```

olikraus, 2016-12-03:
so the solution is simply that the vcom deselect level is set to 0, then the contrast range for 0..255 is much higher.

closed 2016-12-03 by olikraus

p-tric, 2016-12-04:
Great job!

olikraus, 2016-12-04:
Thanks for pointing out this issue. I just took over the vendor suggestions...

olikraus, 2016-12-23:
some OLEDs are not working with the new setup
 * restored init sequence for ssd1306/sh1106 to the value before this issue
 * introduced a new device (VCOMH0) with the init sequence of this issue
 * see issue #116

limpkin, 2018-01-01:
for everyone stumbling on this issue like I did, and who would like to know more about the subject: https://www.osram-os.com/Graphics/XPic2/00032223_0.pdf/4-Bit%20Driver%20Basic%20Register%20Setup.pdf

Wookbert, 2022-03-24:
So we have an 1.3" SH1106 OLED, 4 pin SPI connected, using SW buffer. PreCharge works, but we just can't properly reduce the brightness using VComSelect. No matter which value we try, the display turns completely black. 0xDB -> display goes black.

What's the trick!?

bmoerdyk2, 2022-06-14:
Display: https://www.waveshare.com/wiki/2.23inch_OLED_HAT.  Using I2C mode, as I have with another display (based on SSD1306) that I have used.
SSD1305 with 128x32 OLED display.  Setting VCom to 0 with high contrast levels results in vertical lines on the display, of varying brightness.  Setting VCom limit to 10 and precharge to 114 is enough to remove them, but seems to also drastically limit the minimum dimming level.
Anyone else have this experience?
I'm supplying 3.3V to VDD on the display. (datasheet says it can be 5V, but I haven't tried it yet.)

olikraus, 2022-06-15:
I have created an example code here which may be used as an stating point:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/SSD1306DimToZero/SSD1306DimToZero.ino

VCom value 0 might lead to an instable display behavior. VCom value 1 often works.

To reach lowest possible values, use a set contrast value of 0 (or maybe 1).
Then further lower the precharge value p2 (see the above mentioned example).

bmoerdyk2, 2022-06-15:
Excellent.  Thank you.