olikraus/u8g2 #101: UC1608 240X128 Constructor

GavinBrownAus, 2016-12-06:
Hi Oliver,

Hope your well?

Really like using your older U8G library on Arduino - its excellent. Have recently migrated onto the Teensy boards with a UC1608 controller and its even better - thank you. 

I'm looking to migrate across to your new U8G2 library with the Teensy also using the UC1608 controller. 

Can you advise how I go about porting the following constructor please?

U8GLIB_UC1608_240X128 u8g(2, 3, 4, 5);

I can see the 240X64 controller but not the 240X128 version. 

Any help is greatly appreciated.

olikraus, 2016-12-06:
Can you provide the data sheet?

GavinBrownAus, 2016-12-06:
Please see attached. 
[RX240128A-GHW.pdf](https://github.com/olikraus/u8g2/files/634995/RX240128A-GHW.pdf)

olikraus, 2016-12-06:
Can you test the new UC1608_240X128 constructor?
Please download U8g2lib 2.8.1 from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
You need to add this zip to your Arduino IDE (Add Zip Library).

Thanks for testing

GavinBrownAus, 2016-12-07:
Hi Oliver,

Went through all of the UC240x128 contructors and none of them worked.

The SW SPI versions didnt work, nor the HW ones (as expected) didnt work as
its not a HW setup.

The following is the constructor from U8G if that helps?

I use a 4 pin SW mode, with param 1 - SCK, 2 - MOSI, 3 - CS, 4 - A0

I dont see that combination in the U8G2 constructors.

Am I missing somehting maybe?

Regards

Gavin Brown

class U8GLIB_UC1608_240X128 : public U8GLIB
{
  public:
    U8GLIB_UC1608_240X128(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t
a0, uint8_t reset = U8G_PIN_NONE)
      : U8GLIB(&u8g_dev_uc1608_240x128_sw_spi, sck, mosi, cs, a0, reset)
      { }
    U8GLIB_UC1608_240X128(uint8_t cs, uint8_t a0, uint8_t reset =
U8G_PIN_NONE)
      : U8GLIB(&u8g_dev_uc1608_240x128_hw_spi, cs, a0, reset)
      { }
};

class U8GLIB_UC1608_240X128_2X : public U8GLIB
{
  public:
    U8GLIB_UC1608_240X128_2X(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t
a0, uint8_t reset = U8G_PIN_NONE)
      : U8GLIB(&u8g_dev_uc1608_240x128_2x_sw_spi, sck, mosi, cs, a0, reset)
      { }
    U8GLIB_UC1608_240X128_2X(uint8_t cs, uint8_t a0, uint8_t reset =
U8G_PIN_NONE)
      : U8GLIB(&u8g_dev_uc1608_240x128_2x_hw_spi, cs, a0, reset)
      { }
};

olikraus, 2016-12-07:
Can you post the u8glib constructor which worked for you?

GavinBrownAus, 2016-12-07:
This is what worked using U8G.

U8GLIB_UC1608_240X128 u8g(2, 3, 4, 5);

Kind regards

Gavin Brown

olikraus, 2016-12-07:
ok, this means you should use this constructor, right?
U8G2_UC1608_240X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 2, /* data=*/ 3, /* cs=*/ 4, /* dc=*/ 5, /* reset=*/ U8X8_PIN_NONE);

Yet, this is not working, correct?

GavinBrownAus, 2016-12-07:
Yep tried that one, but it doesnt work either.

I did tell you I was using a Teensy 3.2 with the display?

Kind regards

GB

olikraus, 2016-12-07:
ok, what about
U8G2_UC1608_ERC24064_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 2, / data=/ 3, / cs=/ 4, / dc=/ 5, / reset=*/ U8X8_PIN_NONE);

Will this show anything?

GavinBrownAus, 2016-12-08:
Hi Oliver,

Nope, nothing for that one.

U8G2_UC1608_ERC24064_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 2, / data=/ 3, /
cs=/ 4, / dc=/ 5, / reset=*/ U8X8_PIN_NONE);

However, the following constructor:

U8G2_UC1608_240X128_F_3W_SW_SPI  u8g2(U8G2_R0, 2, 3, 4, 5);

Gives me sqiggly lines on the screen and random weird looking text on the
screen.

Hope this helps.

Regards

GB

olikraus, 2016-12-08:
Seeing at least something is always a good news. Can you post a picture how it looks like?
Thanks...

GavinBrownAus, 2016-12-08:
Hi Oliver,

Photo attached.

Hope this helps.

Kind regards

Gavin Brown

[image: Inline images 1]

olikraus, 2016-12-08:
somehow i am not able to see the image :-(

GavinBrownAus, 2016-12-08:
How about this time?

Regards

GB

olikraus, 2016-12-08:
no :-(
Can you e-mail this to me?
Strange...

olikraus, 2016-12-08:
ok,got it. Which example did you use? HelloWorld.ino?

GavinBrownAus, 2016-12-08:
Yep, Hello  World as below:

GavinBrownAus, 2016-12-08:
Yep, Hello World as below code:

#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

U8G2_UC1608_240X128_F_3W_SW_SPI  u8g2(U8G2_R0, 2, 3, 4, 5);

void setup() {
  // put your setup code here, to run once:

  u8g2.begin();
}

void loop() {
  // put your main code here, to run repeatedly:

u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(10,15,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);

}

olikraus, 2016-12-08:
ok, thanks a lot so far.
Can you check the latest release from here: 
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Thanks

GavinBrownAus, 2016-12-09:
Hi Oliver,

Success!!!

The new library with the following constructor works:

U8G2_UC1608_240X128_F_4W_SW_SPI  u8g2(U8G2_R0, 2, 3, 4, 5);

Thank you for your amazingly fast response in sorting out the issues.

Board: Teensy 3.2
LCD: Raystar RX240128A-GHW

Onward and upward with the new U8G2 library!

Kind regards

Gavin Brown

olikraus, 2016-12-09:
Great. Thanks for feedback and testing.

closed 2016-12-09 by olikraus