olikraus/u8g2 #102: st7565_nhd_c12864 port

dingdinglhz, 2016-12-08:
I have a dev board with Atmel SAM3U2 and nhd_c12864 display, using st7565 controller. I want to use it with u8g2 because I want a full size buffer. But as I can see it is currently unavailable u8glib, although u8glib had this driver. How should I go about porting it?

olikraus, 2016-12-08:
What happens when using one of the existing devices?
How does it look? What is wrong?

dingdinglhz, 2016-12-09:
Sorry but I am not quite familiar with programming in ARM and just started learning. 
But the wiki shows the setup function `u8g2_Setup_st7565_ea_dogm128_f(u8g2, rotation, u8x8_byte_4wire_sw_spi, uC specific)` or `U8G2_ST7565_EA_DOGM128_F_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])` and I assume I need to either provide the pin numbers, but I am not sure how to do that in a custom ARM dev board; or I need to construct a `u8x8_byte_4wire_sw_spi` interface for SPI communication, which I also could not find how to do. Maybe a bit more documentation or examples could help?
Thanks. I apologize for asking any stupid questions.

olikraus, 2016-12-09:
Oh, you don't have an Arduino environment?
Yes, sure this documentation is missing.

olikraus, 2016-12-09:
I have one example here:

https://github.com/olikraus/lpc11u3x-gps-logger/tree/master/u8g2_logo/src

Especially look at u8x8_lpc11u3x.c

closed 2016-12-11 by olikraus

CrayCrayy, 2017-02-09:
Hi olikraus,

Firstly, thank you for all your hard work with this library - it's fantastic.

I am also trying to use an st7565_nhd_c12864, and the st7565_nhd_c12832 constructor works perfectly except for the code not knowing that there are actually 64 pixels instead of 32 in the vertical column. I can even write to the 4th to 7th "tiles" on the lower half of the screen. However, u8x8.begin() does not clear the lower half of the screen.

I have tried porting the st7565_nhd_c12864 using the st7565_nhd_c12832  code as a base, following the new display tutorial you wrote. However I have just started learning C++, and playing with Arduinos, and do not know how to do the make build step. I am also getting this compile error:

`error: 'u8x8_d_st7565_nhd_c12864' was not declared in this scope`

which I traced to my addition of

`u8x8_Setup(getU8x8(), u8x8_d_st7565_nhd_c12864, u8x8_cad_001, u8x8_byte_arduino_4wire_sw_spi, u8x8_gpio_and_delay_arduino);` in the U8x8lib.h file - see below for the code I added

I have updated the header and c files by adding the following directly underneath the original:
in U8x8lib.h

`class U8X8_ST7565_NHD_C12864_4W_SW_SPI : public U8X8 {
  public: U8X8_ST7565_NHD_C12864_4W_SW_SPI(uint8_t clock, uint8_t data, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8X8() {
    u8x8_Setup(getU8x8(), u8x8_d_st7565_nhd_c12864, u8x8_cad_001, u8x8_byte_arduino_4wire_sw_spi, u8x8_gpio_and_delay_arduino); // **This line is causing the compiler to throw the error**
    u8x8_SetPin_4Wire_SW_SPI(getU8x8(), clock, data, cs, dc, reset);
  }
};`

in u8x8_d_st7565.c - copied the 128632 definitions, renaming the section /* NHD-C12864 */ and changed:
1 -  `/* pixel_height = */` from 32 to 64,
2 - `static const uint8_t u8x8_d_st7565_nhd_c12832_init_seq[] = {` to `static const uint8_t u8x8_d_st7565_nhd_c12864_init_seq[] = {`
3 - `uint8_t u8x8_d_st7565_nhd_c12832` to `uint8_t u8x8_d_st7565_nhd_c12832`
4 - `&u8x8_st7565_128x32_display_info` to `&u8x8_st7565_128x64_display_info` in the first case
5 - `u8x8_d_st7565_nhd_c12832_init_seq` to `u8x8_d_st7565_nhd_c12864_init_seq` in the second case

Do I need to do the make build step to get it working, or did I miss editing a "32" to "64" somewhere? I can't find any other code that has c12832 references in it, and thought that since I already added the definition in the .c file, it should not be telling me I did not declare it?

Thank you