olikraus/u8g2 #103: 400KHz Option for ssd1306

edeleflie, 2016-12-09:
I've just moved from Adafruit's libs to u8g2, and noticed that the display refresh is much slower. From a little bit of googling, it seems that u8g2 uses a refresh rate of 100Khz, much lower than the 400khz capable by the 128x64 pixel SSD1306 screens I use.

u8g has a U8G_I2C_OPT_FAST setting ... but I cant find that in u8g2 ... is it hidden somewhere, or perhaps not implemented?

olikraus, 2016-12-09:
Changing the i2c speed is not documented for wire.h.
Yet there is wire.setClock(f).  f seems to be frequency in Herz. Not sure when this was introduced.

olikraus, 2016-12-09:
Update: this is already implemented. Which board do you use? What is your u8g2 version?

edeleflie, 2016-12-09:
Hi Oli,

Thanks for your fast response. I'm loving the fonts compared to Adafruit's
version.

I'm using a new version of Arduino Mega, using u8g2 version downloaded off
Arduino's library manager about 24 hours ago.

I'm using full-frame bufffering ... U8G2_SSD1306_128X64_NONAME_F_SW_I2C

My code looks like this:

  OLEDScreen.clearBuffer();
  OLEDScreen.setFont(u8g2_font_etl14thai_t);
  OLEDScreen.drawStr(36,0,"< speed >");
  OLEDScreen.drawStr(39,28,"/ dist \\");
  OLEDScreen.drawStr(50,18,"+");
  sprintf (buf, "%d", rotary_increment);
  OLEDScreen.drawStr(55,18, buf );
  OLEDScreen.setFont(u8g2_font_etl14thai_t);
  sprintf (buf, "%d", setting_right_wheel_distance);
  OLEDScreen.drawStr(66, 46, buf) ;
  OLEDScreen.sendBuffer();

I'm not setting the Khz rate anywhere ... I'm guessing there is an API call
somewhere or it, but I cant find it.

(I need to draw lines and squares too)

Etienne

olikraus, 2016-12-10:
Etienne, setClock() is used in line 510 of U8x8lib.cpp:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp

In u8glib i manually wrote to the AVR hardware registers, however now i want to be more portable, so i just call the Wire function to set the speed.
If it is still not working, then please check the Arduino version of your IDE. It should be 1.6.0 or higher. Moreover the setClock procedure is not supported an all boards.

edeleflie, 2016-12-10:
Hi Oli,

I'm using v1.6.9 of Arduino, (v2.7.5 for u8g2)

There's something I'm not understanding... I'm using u8g2 not u8x8 ... does
setClock()  still get called? I just tried text in u8x8, and it is much
faster, but still no where near as fast as what I was getting with Adafruit
libraries.

I'm assuming my board supports the faster rate simply because the Adafruit
SSD1306 (and GFX) libs were so fast.

I have a rotary encoder, whose clicks only get read if the screen refresh
is fast enough so that the changes in voltage get read by the Arduino.

Etienne

olikraus, 2016-12-10:
Oh, i see the problem. You should use hw i2c:

U8G2_SSD1306_128X64_NONAME_F_HW_I2C

edeleflie, 2016-12-10:
Hi Oli,

Ah, that's it. Well seen!

Thanks very much for your time. (Now I'll go and research how software I2C
works!)

Etienne

olikraus, 2016-12-10:
SW I2C is just a software side emulation of the i2c hardware, which is of course very slow. However, it works with any digital pin.

olikraus, 2016-12-10:
Can i close this issue?

edeleflie, 2016-12-10:
>
> Can i close this issue?

yes. And thanks again for all the fonts!

Etienne

> <https://github.com/olikraus/u8g2/issues/103#issuecomment-266243226>, or mute
> the thread

olikraus, 2016-12-10:
no problem....

closed 2016-12-10 by olikraus

cyberman54, 2018-07-11:
@olikraus i ran into issues with clock speed 400khz using latest arduino-Espressif32 framework on PlatformIO, since with this upgrade arduino version changed from 1.5.3 to 1.6.0. This triggered speed change from 100khz to 400khz with u8x8, so far as i see.

How can i set a fixed 100khz i2c speed in u8x8, using a SSD1306 OLED controller on HW i2c bus?

cyberman54, 2018-07-12:
Currently and since Espressif32 core version 1.1.x there is an interrupt issue with i2c bus, probably affecting Wire(), that causes bus errors depending on bus speed. That's why i need to reduce bus speed to 100khz until this issue is solved. Detail see link below.

Can i change bus speed with u8g2 during runtime / initialization, or do i need to change it in code and compile the lib new?

https://github.com/espressif/arduino-esp32/issues/1588

olikraus, 2018-07-12:
You could change the value in the code from 4 to 1 and recompile everything, but I do not know whether this value is considered in the u8g2 hal code. The u8g2 hal code was not written by me.