olikraus/u8g2 #106: Bugged drawBitmap function (2.8.4)

viti95, 2016-12-12:
Hi! I think there's something wrong on the last update of the lib (2.8.4) when using the drawBitmap function with an Arduino Mega 2560. It's drawing wrong the bitmaps. This is an example of the function usage:

U8G2_ST7920_128X64_F_HW_SPI u8g(U8G2_R2, 10);

...

const unsigned char grafico9[48] = {
  0x03, 0x00, 0x00, 0x00,
  0x07, 0x80, 0x00, 0x00,
  0x1f, 0xe0, 0x00, 0x00,
  0x3f, 0xf8, 0x00, 0x00,
  0x7f, 0xfe, 0x00, 0x00,
  0xff, 0xff, 0x80, 0x00,
  0x07, 0xff, 0xf0, 0x00,
  0xf0, 0x7f, 0xfc, 0x00,
  0xff, 0x07, 0xff, 0x00,
  0xff, 0xf0, 0x7f, 0xc0,
  0xff, 0xff, 0x07, 0xe0,
  0x1f, 0xff, 0xf0, 0xe0,
};

...

if (rpm >= 9)
    u8g.drawBitmap(6, 9, 4, 12, grafico9);

...

And this is the same code running on 2.7.5 and 2.8.4:

https://youtu.be/sGgVgdKVZ4s
https://youtu.be/UcWEsYUTw_w

olikraus, 2016-12-12:
Can you locate u8g2.h?
Can you comment out line 81 in u8g2.h:
//#define U8G2_WITH_ONE_PIXEL_OPTIMIZATION
Will this fix the problem?

olikraus, 2016-12-12:
I have created a bugfix release 2.8.5.
It should be available soon.
Thanks for this!

closed 2016-12-12 by olikraus

viti95, 2016-12-12:
Wow that was fast!! Thanks!!

olikraus, 2016-12-12:
:-)