olikraus/u8g2 #109: Update to new library KS0108 128 * 64

davixete, 2016-12-16:
Hello
Forgive my English.

I am updating to the new library (U8G2). With Arduino MEGA. My previous configuration was (U8GLIB):

U8GLIB_KS0108_128 u8g (22, 23, 24, 25, 26, 27, 28, 29, 37, 33, 34, 36, 35);
// 8Bit Com: D0..D7: 22,23,24,25,26,27,28,29 in = 37, cs1 = 33, cs2 = 34, di = 36, rw = 35

It works well.

But with u8g2:

U8G2_KS0108_128X64_1 u8g2 (U8G2_R0, 22,23,24,25,26,27,28,29, / * enable = * / 37, / * dc = * / 36, / * cs0 = * / 34, / * cs1 = * / 33, / * cs2 = * / U8X8_PIN_NONE, / * reset = * / 35); // Set R / W to low!

It shows nothing.

LCD Brand: ERM12864-2E - EASTRISING TECHNOLOGY

Thank you very much for your help.

olikraus, 2016-12-16:
R/W is not the reset pin. You have to pull pin 35 to low. Maybe use digitalWrite for this. Leave the reset pin unconnected for the new constructor U8X8_PIN_NONE

davixete, 2016-12-16:
Hello oliver

Thank you very much for your reply. I have tried the code:

U8G2_KS0108_128X64_1 u8g2 (U8G2_R0, 22,23,24,25,26,27,28,29, / *enable = */ 37, /* dc = */ 36, /* cs0 = */ 34, /* cs1 = */ 33, /* cs2 = */ U8X8_PIN_NONE, /* reset = */ U8X8_PIN_NONE); // Set R / W to low!

The GLCD does not work. Does not display image. It does not show compilation errors.

Thank you very much for your help.

olikraus, 2016-12-16:
If u8glib works, than also u8glib should work and it is only a question if the correct constructor setup. How did you solve the r/W problem?

davixete, 2016-12-17:
Hello oliver
I think I understand. I have set pin 35 to low.
My example:

`
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
//arduino mega
U8G2_KS0108_128X64_1 u8g2 (U8G2_R0, 22,23,24,25,26,27,28,29, /* enable = */ 37, /* dc = */ 36, /* cs0 = */ 34, /* cs1 = */ 33, /* cs2 = */ U8X8_PIN_NONE, /* reset = */ U8X8_PIN_NONE);   // Set R/W to low!
const int resetPin = 35; // pin r/w

void setup(void) {
  pinMode(resetPin,OUTPUT);
  u8g2.begin();
  digitalWrite(resetPin, LOW);
}

void loop(void) {
  
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB14_tr);	// choose a suitable font
  u8g2.drawStr(0,20,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
   delay(1000);
}
`
Does not work.
The pin r / w should always be in low ?. I read that low is for writing. High for reading. it is right?

olikraus, 2016-12-17:
I think there was a mistake in the example. Instead of
U8G2_KS0108_128X64_1
it should be
U8G2_KS0108_128X64_F

Oliver

davixete, 2016-12-17:
Great !. It works well.
Oliver, thank you very much for your help. You do a great job.
Thank you!!

olikraus, 2016-12-17:
thanks, I am glad that this finally works.

closed 2016-12-17 by olikraus