olikraus/u8g2 #110: u8g2.print(string); won't work.

FrostMannen, 2016-12-17:
I have tried this code snipp from another issue from aug. Just to try the print statement.

`u8g2.firstPage();`
`do {`
  `//u8g2.setFont(u8g2_font_ncenB14_tr);`
    `    u8g2.setFont(u8g2_font_7x14B_tf);`
    `    u8g2.drawPixel(x,20);`
    `    u8g2.setCursor(10,10);`
    `    u8g2.print(x);`
  `  } while ( u8g2.nextPage() );`
  `  delay(500);`
  `  if(x++>128) x=0;`

The  only thing that is happening to the screen is that it will become blank! But if I comment out the u8g2.print(x); it works again!

Version according to the readme.md: U8glib V2 library for Arduino

FrostMannen, 2016-12-17:
My fault, I had old buggy version. Updated to 2.8.5 and NOW it works like a charm :+1:

olikraus, 2016-12-17:
Great.

closed 2016-12-17 by olikraus