olikraus/u8g2 #112: userInterfaceSelectionList returned values incorrent

nanokatz, 2016-12-18:
Hello;
I've noticed a issue with the use of the userInterfaceSelectionList with arduino.
I can set up the menu fine and navigate no issue.
According to the docs userInterfaceSelectionList will return values of 1-n when the select button is pressed and 0 when the home  button is pressed. Instead it returns values of 0-(n-1) when a selection is made and the home button simply returns the start position specified in the function call.
This seems to be the correct behaviour according the C library documentation.

However this seems to defeat the hole purpose of a home/cancel button as it requires that you have "back/home/cancel" item in the list and that it the initial selection.
Am I misunderstanding the issue here?

Some relavent parts of my code (not in the actually functions of course)
`
const char *main_list =
  "Living Room\nBedroom\nBedtime\nWatching TV\nGoing Out!";
U8G2_ST7920_128X64_2_SW_SPI u8g2(U8G2_R0, /* clock=*/ 4, /* data=*/ 3, /* CS=*/ 2);

u8g2.begin(/*Select=*/ A4, /*Right/Next=*/ A2, /*Left/Prev=*/ A1, /*UP*/ A5, /*DOWN*/ A3, /*home*/ A0);`

current_selection = u8g2.userInterfaceSelectionList("Select Room", 1, main_list);

olikraus, 2016-12-18:
so you think it is better to return 0 here? In fact this is also the behavior for the u8x8 selection list...

nanokatz, 2016-12-18:
Well I am just confused and have not used the u8x8 functions at all. The reference on github says the following about it:
> Returns: 1 to n for if one of the buttons had been selected. 0 if the user has pressed the home/cancel button.

However the C library used by the arduino code says the following
>   returns start_pos if user has pressed the home key
  returns the selected line if user has pressed the select key

So far I am only seeing the the second behaviour in my arduino code. So either the documentation on github is incorrect or there is something I have missed with the setup.
Personally I would find the version where 0 is used only for cancel much more useful and flexible; but hey either way works

olikraus, 2016-12-18:
the description on the wiki page is wrong. 
values 0..n-1 are returned for the selected element. 
I guess the problem was, that i took over the documentation from u8x8 procedure, which indeed works differently.
So i currently think to modify the code instead of the documentation.

nanokatz, 2016-12-18:
Ah okay great. I made a slight change in the library and it works as expected now.

olikraus, 2016-12-18:
ok, but i will keep this open until i have modified the code.
ToDo: takover behavior from u8x8 selection list procedure

olikraus, 2016-12-20:
done, selection list now behaves as described in the manual.

closed 2016-12-20 by olikraus