olikraus/u8g2 #113: ssd1309

olikraus, 2016-12-20:
ordered...

$ more 2.42inch_oled_with_ssd1309.txt 
Diagonal Screen Size：2.42"
Resolution: 128 * 64
Driver IC: SSD1309
SPI interface definition:
1.GND (power ground)
2.VCC (power supply positive)
3.SCL (clock line)
4.SDA (data line)
5.RES (reset line)
6.DC (data / command)
7.CS (chip select)

Seventh leg leads CS signal default; if not, you can go at this time R18 welding CS signal directly to ground. Operation is very

Convenience.

Interface Selection: to choose SPI or IIC interface through R15, R17, R13
This module defaults to the SPI interface, R17 welding, R15, R13 is not welding.

Package Contents
100% Brand New

1 x2.42" 12864 OLED IIC I2C SPI OLED Display Module

hoss183, 2017-01-11:
also requesting SSD1309. I willing to help test or supply parts if required.

olikraus, 2017-01-11:
Thanks a lot. I still hope that this display finds the way to my lab.

olikraus, 2017-01-11:
I have started with the implementation. A first U8X8 constructor is there:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Add the zip lib file in the library manager of the Arduino IDE.

olikraus, 2017-01-13:
done

closed 2017-01-13 by olikraus

hoss183, 2017-01-16:
I'm just testing this, but get this compile error:
Error compiling for board Arduino/Genuino Uno
Build failed for project 'ISU'
ccO8oWXd.ltrans1.ltrans.o*: In function global constructors keyed to 65535_0_ISU.cpp.o.9315
ccO8oWXd.ltrans1.o*: (.text.startup+0xaa): undefined reference to u8g2_Setup_ssd1309_128x64_noname_1
 
collect2.exe*: error: ld returned 1 exit status

with constructor:
U8G2_SSD1309_128X64_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 2, /* data=*/ 3, /* cs=*/ 4, /* dc=*/ 5, /* reset=*/ U8X8_PIN_NONE);  

The error does not mean much to me, any ideas?

olikraus, 2017-01-16:
Which version do you used? The one which is there in the Arduino IDE? It should be 2.12.6
Please do not use anything from the gitbub dir at the moment.

Maybe you need to remove u8g2 manually from your install directory and then reinstall u8g2.

hoss183, 2017-01-16:
I'm using 2.12.6 installed by the Arduino IDE. I did actually delete the library manually before installing again.

olikraus, 2017-01-16:
I have created a new issue for this --> #162