olikraus/u8g2 #118: U8G2 doesnt play nicely with adafruit TSL2561

scrumfled, 2016-12-23:
Previously working fine with a SSD1306 OLED and u8glib, together with teh adafruit libraries for TSL2561. Porting to U8G2 causes the display to stop working. I've got it working in isolation with the OLED, but as soon as the adafruit libs are included it ceases working.

Constructors:

u8glib: U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE | U8G_I2C_OPT_DEV_0);

u8g2: U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE); 

Adafruit Libs:

https://github.com/adafruit/Adafruit_TSL2561/archive/master.zip
https://github.com/adafruit/Adafruit_Sensor/archive/master.zip

olikraus, 2016-12-23:
I guess you need to use HW I2C:
u8g2: U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, reset=*/ U8X8_PIN_NONE);

scrumfled, 2016-12-27:
Thats erm... *shuffles feet*... embarassing ;)

thanks for the reply when you should've been celebrating

closed 2017-01-08 by olikraus