olikraus/u8g2 #119: Amazon Screen Insanity

iotexpert, 2016-12-24:
I bought three I2C displays from Amazon.
[Display 1](https://www.amazon.com/gp/product/B01HI42NPK/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)
[Display 2](https://www.amazon.com/gp/product/B01HHOAQ5A/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)
[Display 3](https://www.amazon.com/gp/product/B01G6SAWNY/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)

All three were purported to be SSD1306.

You should be aware that 
1. The VCC and GND lines are flipped 
2. (display 1) The larger screen seems to not react exactly the same (there are wild pixels on the right side)
3. (display 3) has yellow at the top (which I should have noticed) and also seems to have gaps between the pixels.

I was using u8x8 with SSD1306 Noname

Alan Hawse
engineer@iotexpert.com
www.iotexpert.com

olikraus, 2016-12-24:
Because this is the issue list for U8g2, my question is: Is there any issue with u8g2? Otherwise i would like to close this topic.

The description for display 1 mentions controller SSD1106. However such a controller does not exist. I guess it is a writing error and means SH1106. If you would use the SH1106 constructor from u8g2, then the pixel issue will be fixed.

Bascially i can not help in any way for wrong product descriptions or buggy products...

If you want to inform people, it is probably better to leave a review in amazon itself.

iotexpert, 2016-12-24:
I think that you should close this topic.  I only put it here to keep other people out of trouble.

olikraus, 2016-12-24:
Great, thanks for sharing.

closed 2016-12-24 by olikraus