olikraus/u8g2 #120: Porting Guide (a start)

iotexpert, 2016-12-24:
If you want to use another MCU (a PSoC in my case) it would be best to implement at least two custom functions 
1. The I2C hardware interface  (e.g. u8x8_byte_hw_i2c)
2. The GPIO and delay interface function (e.g. psoc_gpio_and_delay_cb)

These functions are both used as callbacks by the U8X8 library.  They are setup when you call the first initialization function e.g.
u8x8_Setup(&u8x8, u8x8_d_ssd1306_128x64_noname, u8x8_cad_ssd13xx_i2c, u8x8_byte_hw_i2c, psoc_gpio_and_delay_cb);

**(1) I2C Hardware Interface**
The I2C HW Interface function (in fact all of the HAL function) .  The function must conform with the function prototype:
 
typedef uint8_t (*u8x8_msg_cb)(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
 
[editorial note: I think that the msg should be an enumerated datatype).

The HW interface function needs to handle message from the rest of the system.  The messages that seem to matter are:
U8X8_MSG_BYTE_SEND
U8X8_MSG_BYTE_INIT
U8X8_MSG_BYTE_END_TRANSFER
U8X8_MSG_BYTE_START_TRANSFER
U8X8_MSG_BYTE_SET_DC

Here is an example for a Cypress PSoC:
/* 
 * A PSoC I2C Hardware Interface for U8X8
*/

uint8_t u8x8_byte_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    #ifdef DEBUG_UART
    char buff[20];
    #endif
    uint8_t *data;
    switch(msg)
    {
        case U8X8_MSG_BYTE_SEND:
            data = (uint8_t *)arg_ptr;
            while( arg_int > 0 )
            {
                #ifdef DEBUG_UART
                sprintf(buff,"%x\n",*data);
                UART_UartPutString(buff);
                #endif
                (void)I2C_I2CMasterWriteByte(*data); // No Error Checking..
  	            data++;
	            arg_int--;
            }
        break;
      
        case U8X8_MSG_BYTE_INIT: // Using the HW block you dont need to set 1/1
        break;
        case U8X8_MSG_BYTE_SET_DC: // not used... I2C only
        break;
        case U8X8_MSG_BYTE_START_TRANSFER:
            (void)I2C_I2CMasterSendStart(u8x8_GetI2CAddress(u8x8)>>1,I2C_I2C_WRITE_XFER_MODE); // no Error Checking
         break;
        
        case U8X8_MSG_BYTE_END_TRANSFER:
            (void)I2C_I2CMasterSendStop(); // no error checking
        break;
    
        default:
            return 0;
    }
    
    return 1;
}

**(2) GPIO and Delay Interface**
The I2C HW Interface function (in fact all of the HAL function) .  The function must conform with the function prototype:
 
typedef uint8_t (*u8x8_msg_cb)(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

This function is used to set and reset GPIOs (in the case if software implemented interfaces) e.g. a software I2C, SPI, 8080 or 6800 interface.  In addition this function is used to implement busy-wait delays for pin timing.

For my implementation I used a hardware block for the I2C interface... so I didnt implement any of the GPIO functions.  HOWEVER, to do that you would just need to write the correct pin to the correct state e.g. for PSoC Pin U8X8_RESET_PIN_Write(arg_int);

The delay functions provide timing for software pin writes.  You need to implement a busy-wait delay for each of those messages.  I used the Cypress provided delay functions. i.e. CyDelay

/* 
 * A PSoC implementation of the delay and gpio callback:
 * 
*/
uint8_t psoc_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    (void)u8x8;
    (void)arg_ptr;
    switch(msg)
    {
        case U8X8_MSG_GPIO_AND_DELAY_INIT: // for now who cares?
        break;
        case U8X8_MSG_DELAY_MILLI:
            CyDelay(arg_int);
        break;
        case U8X8_MSG_DELAY_10MICRO:
            CyDelayUs(10);
        break;
        case U8X8_MSG_DELAY_100NANO:
            CyDelayCycles((CYDEV_BCLK__SYSCLK__HZ/1000000) * 100 - 16); //PSoC system reference guide says ~16 extra cycles 
        break;
        /*    - My Display has only I2C... which I have implemented in HW
         * If you want to use a software interface or have these pins then you 
         * need to read and write them
        case U8X8_MSG_GPIO_SPI_CLOCK:
        case U8X8_MSG_GPIO_SPI_DATA:
        case U8X8_MSG_GPIO_CS:
        case U8X8_MSG_GPIO_DC:
        case U8X8_MSG_GPIO_RESET:
        case U8X8_MSG_GPIO_SPI_CLOCK:
        case U8X8_MSG_GPIO_SPI_DATA:		
        case U8X8_MSG_GPIO_D0:    
        case U8X8_MSG_GPIO_D1:			
        case U8X8_MSG_GPIO_D2:		
        case U8X8_MSG_GPIO_D3:		
        case U8X8_MSG_GPIO_D4:	
        case U8X8_MSG_GPIO_D5:	
        case U8X8_MSG_GPIO_D6:	
        case U8X8_MSG_GPIO_D7:
        case U8X8_MSG_GPIO_E: 	
        case U8X8_MSG_GPIO_I2C_CLOCK:
        case U8X8_MSG_GPIO_I2C_DATA:
        break;  */
    }
    return 1;
}

olikraus, 2016-12-24:
Thanks for shareing this information, no doubt it will stay here, but i have to close this issue sooner or later. Maybe it is a better idea to setup a web page with this information somewhere.

iotexpert, 2016-12-24:
Yes.  You should close this issue.  I only put it here for other people.  Actually if you will accept a pull request I will put it into a file in the doc directory... or in whatever you envision your documentation to be going forward.

olikraus, 2016-12-24:
All my docs are in u8g2.wiki. Not sure whether pull requests are possible on the wiki part, but yes, would accept this.

iotexpert, 2016-12-24:
OK.  Ill write the porting guide (for at least the C-version).... then Ill do a pull request.  Then you can decide what do with the file... either leave it in the doc directory... or wiki or... whenever you want it.  Your choice.  If you decide Wiki... then give me permission and Ill do the documentation there.  Your choice... (I have to go to a Christmas party... so it will be later today or tomorrow).

olikraus, 2016-12-24:
sounds interessting... merry Christmas

olikraus, 2016-12-24:
U8g2 Invitation link: 
https://github.com/olikraus/u8g2/invitations

iotexpert, 2016-12-25:
I just created the first crack at the porting guide page.  You can see it at https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

I want to add the example code inline... but Im not totally sure how to make that look right.  As soon as I fix it Ill update this comment and link it to the main page.

iotexpert, 2016-12-25:
When you are satisfied with what I did please go ahead and close the issue.

olikraus, 2016-12-26:
Nice. Thanks a lot. In fact as a none native speaker I always hasitate to write much in english...
Let us use this issue to track progress on the porting guide. I also want to contribute to that. At least i would like to do a review on it. So maybe you just let me know once i can start the review.

iotexpert, 2016-12-26:
Your english is quite a bit better than my German... though I have spent a good deal of time there this year.

olikraus, 2017-02-15:
ok,i am closing this. Thanks for starting the porting guide. Meanwhile it helped a lot of users.

closed 2017-02-15 by olikraus